evshary/autoware_carla_launch #1: Failed to execute the docker

YuanYuYuan, 2023-05-19:
```bash
./run-docker.sh
```

```bash
Extension volume doesn't support default arguments. Please extend it.
Active extensions ['network', 'nvidia', 'privileged', 'volume', 'x11', 'user']
Step 1/12 : FROM python:3-slim-stretch as detector
 ---> 7691d3cb6cbc
Step 2/12 : RUN mkdir -p /tmp/distrovenv
 ---> Using cache
 ---> 9d7a12f57f51
Step 3/12 : RUN python3 -m venv /tmp/distrovenv
 ---> Using cache
 ---> 533adc7ef328
Step 4/12 : RUN apt-get update && apt-get install -qy patchelf binutils
 ---> Running in 062552a9721d
Ign:1 http://security.debian.org/debian-security stretch/updates InRelease
Ign:2 http://deb.debian.org/debian stretch InRelease
Ign:3 http://security.debian.org/debian-security stretch/updates Release
Ign:4 http://deb.debian.org/debian stretch-updates InRelease
Ign:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:6 http://deb.debian.org/debian stretch Release
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:8 http://deb.debian.org/debian stretch-updates Release
Ign:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Err:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
  404  Not Found
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:7 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Err:10 http://deb.debian.org/debian stretch/main amd64 Packages
  404  Not Found
Ign:11 http://deb.debian.org/debian stretch-updates/main all Packages
Err:12 http://deb.debian.org/debian stretch-updates/main amd64 Packages
  404  Not Found
Reading package lists...
[91mW: The repository 'http://security.debian.org/debian-security stretch/updates Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch-updates Release' does not have a Release file.
E: Failed to fetch http://security.debian.org/debian-security/dists/stretch/updates/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://deb.debian.org/debian/dists/stretch/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://deb.debian.org/debian/dists/stretch-updates/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
[0m
Removing intermediate container 062552a9721d
no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'autoware-carla-bridge', maybe the base image does not exist
```

evshary, 2023-05-22:
I believe this issue is related to rocker.
Here is the workaround:
https://github.com/orgs/autowarefoundation/discussions/3470#discussioncomment-5759471

YuanYuYuan, 2023-05-22:
That solves my problem.😁

 Thanks a lot!

closed 2023-05-22 by YuanYuYuan