evshary/autoware_carla_launch #3: Error while building container for carla bridge

Kim-mins, 2023-06-11:
Hi @evshary! Thank you for the nice work!
I saw your comment [here](https://github.com/orgs/autowarefoundation/discussions/3348#discussioncomment-6087225) and I'm currently following the instruction for **humble** branch of this repository.
However, while building the container for **carla bridge** (by typing `make prepare_bridge`), I faced an error below:
```shell
...
lactic/lib/x86_64-linux-gnu:/opt/ros/galactic/lib:/usr/local/nvidia/lib:/usr/local/nvidia/lib64 ./python -E -m ensurepip \
                $ensurepip --root=/ ; \
fi
Looking in links: /tmp/tmpx4c31f6u
Requirement already satisfied: setuptools in /home/mins/autoware_carla_launch/pyenv/versions/3.8.10/lib/python3.8/site-packages (56.0.0)
Requirement already satisfied: pip in /home/mins/autoware_carla_launch/pyenv/versions/3.8.10/lib/python3.8/site-packages (21.1.1)
make[1]: Leaving directory '/tmp/python-build.20230611162332.223/Python-3.8.10'
Installed Python-3.8.10 to /home/mins/autoware_carla_launch/pyenv/versions/3.8.10
./script/dependency_install.sh: line 19: cd: external/zenoh_carla_bridge/carla_agent: No such file or directory
make: *** [Makefile:16: prepare_bridge] Error 1
```

I want to ask you some help to solve this error..
My **OS is ubuntu 20.04**, and I'm planning to use **carla 0.9.13**, so it seems there's no violation to the prerequisite..

Thank you in advance! I'm really appreciate to your work, since I could not have any expectation to run autoware universe humble with carla.. thank you so much.

Kim-mins, 2023-06-11:
I just found the solution:
Run `git submodule update --init --recursive` before the preparation.
I did not know there are submodules.

closed 2023-06-11 by Kim-mins

evshary, 2023-06-12:
Thanks for your report! I think this is order issue.
I should run `git submodule update --init --recursive` before installing pyenv
https://github.com/evshary/autoware_carla_launch/blob/humble/Makefile#L17

Kim-mins, 2023-06-12:
I saw your PR for this issue. Thank you for the update!