evshary/autoware_carla_launch #7: Cannot set destination that requires lane change

Kim-mins, 2023-06-15:
Hi @evshary!
I'm currently using your nice bridge software to run Autoware.
However, I faced another issue that **I cannot set destination that requires lane change**.
[Here](https://drive.google.com/file/d/1j10uosM_aVYpUg1FUm8-wDUaR_mBqXdU/view?usp=sharing) is a video for better understanding of the situation.
The map used in the video is Town05. 
When I set the destination near to the start point, which requires a lane change once, but it seems like the car does not make a correct path.
I thought that this is because there's not enough distance to change lane, so I set the destination a bit far from the starting point, but the situation was the same.
When I make the car drive(click AUTO of OperationMode on RViz), it drives but stucks to the same lane and cannot reach the goal.
I also suffered this situation on Town04 too, and I could not even set the destination(On Town05, '2D Goal Pose' call worked anyway, but on town04, even '2d Goal Pose' does not work for 'lane-changing destination').

Have you ever suffered this issue?
I cannot sure whether this is the problem from Autoware or from bridge, but I write this issue to you because I expect Autoware to be able to handle this kind of somewhat basic case.. so I just want to hear your opinion.

I could reproduce the issue at any point of the map with multi-lane(Town04, Town05), so I expect that you can reproduce it with random spawning point(default option of the bridge)

Thank you for reading!

evshary, 2023-06-16:
I think it might be more related to map. I found that mission_planner fails to set goal.
[image]

I remember Autoware should be able to change lane in AWSIM, so the reason I guess might be changing lane is not available in the map.

evshary, 2023-06-16:
I compare the lanelet2_map.osm between Carla and AWSIM. It seems like AWSIM has dashed lane, so the vehicle is able to change lane. Maybe the lanelet2_map.osm should be modified to fulfill your requirements.
https://github.com/fzi-forschungszentrum-informatik/Lanelet2/blob/master/lanelet2_core/doc/LinestringTagging.md

Kim-mins, 2023-06-16:
Thank you for the detailed investigation! I really appreciate to your hard work!
I also found that the osm file of AWSIM(nishishinjuku) contains more information than my current map like below:
```
<way id="20">
    <nd ref="31558"/>
    ...
    <nd ref="31582"/>
    <tag k="type" v="line_thin"/>
    <tag k="subtype" v="dashed"/>
    <tag k="lane_change" v="yes"/>
    <tag k="width" v="0.000"/>
</way>
```
It would be impossible for me to find the reason.. since I'm novice to Autoware. Thank you so much!

By the way, I checked the link you provided regarding line tagging, but.. actually I do not have any idea for the modification.
I just used the map provided [here](https://github.com/evshary/autoware_carla_launch/blob/main/carla_map/README.md), and I don't know about the conversion process.
Is there any proper way to modify the map? or do you know any tutorials on lanelet2 for me?
I think I can also try to convert the opendrive file to osm, but I have no idea for this.. do you have any recommendation?

Sorry for asking you too many questions, but I don't know any expert like you around me.. please tell me if you have any idea for this. Thank you!

Kim-mins, 2023-06-18:
I change every `way` of `.osm` like below:
```xml
<way id="1000">
    <nd ref="1002" />
    <nd ref="1004"/>
+   <tag k="type" v="line_thin"/>
+   <tag k="subtype" v="dashed"/>
+   <tag k="lane_change" v="yes"/>
+   <tag k="width" v="0.150"/>
</way>
```
but the routing does not work as it was.
Could you please tell me whether I'm doing right?

evshary, 2023-06-19:
I'm afraid I can't give you too many suggestions about editing the map. I have little experience with this.
I think maybe you can try to use JOSM mentioned here. It can open the .osm files.
https://github.com/fzi-forschungszentrum-informatik/Lanelet2/blob/master/lanelet2_maps/README.md
But I'm not familiar with the tool, not sure whether it can change the road type or not.

evshary, 2023-06-21:
I found that there is Vector Map Builder which might fulfill your need
https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-maps/#creating-a-vector-map

Kim-mins, 2023-06-21:
Oh sorry for late reply.
I just found the way to modify the map with JOSM, and I'm manually fixing the map to support lane change.
Thank you for the help! I'll try your another suggestion too!

closed 2023-06-21 by Kim-mins

equantumWang, 2024-12-15:
Hi guys I may find the solution of this proble, see issues below:
[https://github.com/orgs/autowarefoundation/discussions/4577](url)
[https://github.com/hatem-darweesh/op_agent/issues/3](url)