evshary/autoware_carla_launch #8: How to spawn 'autopilot

Kim-mins, 2023-06-16:


Kim-mins, 2023-06-18:
I just found that the bridge terminates after about 150 seconds with the log below:
```bash
 INFO  zenoh_carla_bridge::bridge::vehicle_bridge > Detect a vehicle autopilot
 INFO  zenoh_carla_bridge                         > Actor 1606 created
terminate called after throwing an instance of 'std::runtime_error'
  what():  Responding error from function get_wheel_steer_angle: Actor could not be found in the registry.  Actor Id: 1552
Aborted
```
To my knowledge, this error occurs when the actor(the actor with the id 1552, in this case.) is destroyed from carla simulator.
I've observed that **Carla automatically removes the vehicle itself** if it stops too long, and so I guess the reason of the termination of the bridge is that 1) vehicle stops too long, 2) the vehicle is destroyed by Carla after about 150 seconds and 3) the bridge eventually terminates (due to the destroyed vehicles).

But the other NPC vehicles moves after the termination of the bridge, so it seems like **the bridge is not compatible with autopilot NPC Vehicles**.
This can be the problem of my side, but maybe you can give me some hints for the conclusion.. Thank you in advance!

evshary, 2023-06-19:
Does the issue only happen while adding NPC vehicles?
Maybe you can provide me with a way how to reproduce it. I guess there is something wrong with the bridge.
I'll try to find some time to fix it.

evshary, 2023-06-20:
Thanks for your steps to reproduce the bugs. I'll try it on my side.

evshary, 2023-06-21:
@Kim-mins I think the issue is fixed by #10 . Maybe you can try and give me some feedback.

Kim-mins, 2023-06-21:
Thank you so much @evshary!! I always appreciate to your hard work!
I have little time to check it now, so I think I can run the code a bit later.
I'll mention you after the check! Thank you!

Kim-mins, 2023-06-23:
Thank you @evshary! I just tested the driving with NPC vehicle, and everything works fine.
I really appreciate to your help!!! Thank you!

PR #10 (2023-06-21): Fix npc issue
Fix #8 
Do not bridge NPC vehicles with Zenoh. We add "autoware_" before the role_name of the vehicle to distinguish our vehicle from NPC vehicles.
closed 2023-06-23 by Kim-mins