evshary/autoware_carla_launch #13: Run the latest version of Autoware with the bridge

Kim-mins, 2023-07-04:
Hi evshary!

I've got so many help from you, but I just got stuck while using the latest version of Autoware.
I pull the latest docker image [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/106187605?tag=humble-latest-cuda) because the image contains the recent version of Autoware (committed on June 30), and slightly modified your image build script for `zenoh-autoware` referring [here](https://github.com/autowarefoundation/autoware/blob/main/docker/autoware-universe/Dockerfile).
Building Autoware was successfuly, but I encountered an error while running the command `make build_autoware`

Here's the error I met:
```bash
mins@4b4493b67b85:~/autoware_carla_launch$ make build_autoware
colcon build --symlink-install --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
Starting >>> autoware_carla_launch
Starting >>> autoware_manual_control
Starting >>> carla_pointcloud
Starting >>> carla_sensor_kit_description
Starting >>> carla_sensor_kit_launch
Finished <<< autoware_carla_launch [0.13s]
Finished <<< carla_sensor_kit_launch [0.12s]
Finished <<< carla_sensor_kit_description [0.13s]
--- stderr: carla_pointcloud
In file included from /home/mins/autoware_carla_launch/src/carla_pointcloud/src/carla_pointcloud/carla_pointcloud_interface_node.cpp:2:
/home/mins/autoware_carla_launch/src/carla_pointcloud/include/carla_pointcloud/carla_pointcloud_interface_node.hpp:9:10: fatal error: autoware_auto_control_msgs/msg/ackermann_control_command.hpp: No such file or directory
    9 | #include <autoware_auto_control_msgs/msg/ackermann_control_command.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[3]: *** [CMakeFiles/carla_pointcloud_interface.dir/build.make:76: CMakeFiles/carla_pointcloud_interface.dir/src/carla_pointcloud/carla_pointcloud_interface_node.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:139: CMakeFiles/carla_pointcloud_interface.dir/all] Error 2
gmake[1]: *** [Makefile:146: all] Error 2
---
Failed   <<< carla_pointcloud [0.32s, exited with code 2]
Aborted  <<< autoware_manual_control [0.45s]

Summary: 3 packages finished [0.64s]
  1 package failed: carla_pointcloud
  1 package aborted: autoware_manual_control
  2 packages had stderr output: autoware_manual_control carla_pointcloud
make: *** [Makefile:11: build_autoware] Error 2
```

I do not have that much knowledge on the bridge code, but I remember that the package `autoware_manual_control` is from [your repository](https://github.com/evshary/autoware_manual_control).
I want to ask if you have any idea on this issue.

Thank you in advance!

evshary, 2023-07-05:
Do you source Autoware before running `make build_autoware`?
The image I used has already source Autoware while running the image. However, if you build the Autoware from source, you need to source by yourselves.

Something like:
```shell
source ~/autoware/install/setup.bash
```

By the way, I found that there are some problems while running in the latest Autoware.
1. The current Carla map format is not compatible with the latest Autoware. We're still looking at that.
2. You might face the problem with sensors parameter.

zeidk, 2023-07-05:
@evshary Although I'm not concerned about this issue, I would like to thank you for taking the time to help us :1st_place_medal:

Kim-mins, 2023-07-05:
Thank you for the reply evshary!

I'm currently trying to pull the latest Autoware and build it, but I cannot even run Autoware so I cannot encounter two issues you provided.

Here are my steps:
1. pull the image with the tag `humble-latest-cuda`
2. run `./container/run-autoware-docker.sh`
3. remove `/autoware` and clone `release/2023.06` branch to `/autoware` directory
4. Follow steps here: https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/
5. Follow the instruction in this repository and run `./script/run-autoware.sh` and fail to run.

Actually, I failed to install cuda-related libraries while running `./setup-dev-env.sh` on step 4.
When I forcefully proceed to build and run Autoware, I encounter the errors regarding the sensors.

Sorry for bothering, but could you please tell me **the docker image** you are using and **steps to pull and build the latest Autoware**?

Kim-mins, 2023-07-06:
Thank you evshary!
After following your suggestions, now I can run the latest Autoware.
And I can see the black screen on RViz as below, and maybe I'm suffering the same issue you mentioned above regarding the map.

[image]

I also tried to load the `nishishinjuku` map, but I think it is loaded well as the image below. So maybe the maps of carla have some problems..

[image]

I have no idea on this.. do you have any solution for this issue?

Kim-mins, 2023-07-06:
I don't know if this would help, but I found that, on `Town02` of Carla, the map is partially loaded. 

[image]

Also, it is weird that the `Motion` status in RViz is `MOVING`, since the initial pose is never provided.

evshary, 2023-07-07:
We found that the lanelet2 file lacks `mgrs_code` tag, but still under testing now.

Kim-mins, 2023-07-12:
Any updates on this issue? I'm really looking forward to it.
It seems like the error is from the maps. Can we resolve this issue?

evshary, 2023-07-13:
Unfortunately, no. It seems like `mgrs_code` is not the root cause.
I guess the map format requested by Autoware is changed in the recent update.
We're not familiar with the map, so it might take some time...

Kim-mins, 2023-07-13:
Thank you for the answer. Maybe I should ask for the maps to Autoware team..

Kim-mins, 2023-07-20:
Hi @evshary!
Thanks to your question [here](https://github.com/orgs/autowarefoundation/discussions/3660#discussioncomment-6475886), now I can load the map on the latest version of Autoware.
However, when I run Autoware, I cannot set the initial position of ego vehicle.
Maybe this is because the ego vehicle is moving (I don't know why, but when I run Autoware the state of `Motion` of RViz said the car is moving. You can refer to the image [here](https://github.com/evshary/autoware_carla_launch/issues/13#issuecomment-1623049571)).
After changing the map configuration, didn't you suffer the same issue with me?
If not, could you please tell me the solution?

Thanks in advance!

evshary, 2023-07-21:
@Kim-mins I've just updated the humble branch and I believe now we can run with the latest Autoware (I use [humble-20230715-prebuilt-cuda-arm64](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/109791065?tag=humble-20230715-prebuilt-cuda-arm64))
Since the autoware_launch changed a lot, I decided to use [the fork](https://github.com/evshary/autoware_launch/tree/carla_20230715) by myself.
You can try it and give me some feedback.

Kim-mins, 2023-07-23:
Thank you so much @evshary!!
I'm currently busy for another work, so maybe I can try it a bit later.
I'll test it as soon as possible. Thank you.

Kim-mins, 2023-08-07:
Hi @evshary, sorry for late. I was busy for other works.
I just tried the bridge with the latest version of Autoware, and **everything works fine** for me.
Thank you so much to solve this issue! I really appreciate to your hard work!

Autostone-c, 2023-08-07:
> Hi @evshary, sorry for late. I was busy for other works. I just tried the bridge with the latest version of Autoware, and **everything works fine** for me. Thank you so much to solve this issue! I really appreciate to your hard work!

hi, have you ever  encountered the following problems 
https://github.com/evshary/autoware_carla_launch/issues/23#issuecomment-1667175298

Kim-mins, 2023-08-07:
Hi @Autostone-c, maybe you can check the comment of the issue.

Autostone-c, 2023-08-07:
> Hi @Autostone-c, maybe you can check the comment of the issue.

can you carry out secondary planning？when i set second Goal Pose, it doesn't work

Kim-mins, 2023-08-07:
Well.. I'm only interested in setting the goal once..
As far as I know, I remember that I could set the secondary goal(setting another goal after setting a goal) well.
Maybe the difference comes from the settings. It seems like you are running on different setting with me (I'm running everything on one machine).

Autostone-c, 2023-08-07:
> Well.. I'm only interested in setting the goal once.. As far as I know, I remember that I could set the secondary goal(setting another goal after setting a goal) well. Maybe the difference comes [...]

im also on one machine, but my second goal not work. i will try more

Kim-mins, 2023-08-07:
Oh I didn't know that.
FYI, setting a new goal 1) while driving 2) after reaching the previous goal, doesn't matter. I could set the goal in both cases.
Or maybe you can remove the built image (zenoh carla bridge & autoware) and run it again. It won't take time too much. (< 5 minutes)

evshary, 2023-08-08:
I can also set a new goal after the previous goal.
I remember I met the issue while trying the planning simulation example on the Autoware tutorial before, but failed to reproduce it afterward.
Just FYI

evshary, 2023-08-08:
> Hi @evshary, sorry for late. I was busy for other works. I just tried the bridge with the latest version of Autoware, and **everything works fine** for me. Thank you so much to solve this issue! I really appreciate to your hard work!

Glad this helps :smile:	

<br class="Apple-interchange-newline">

closed 2023-08-08 by Kim-mins