evshary/autoware_carla_launch #15: Seems like the perception module (especially prediction) not work

Kim-mins, 2023-07-12:
Hi evshary!
I'm currently running the Autoware universe on Carla with your bridge, but I got a question regarding the perception module of Autoware.

To my knowledge, Autoware is capable of **predicting the trajectory** of objects(like NPC Vehicle, ...) moving around, and I remember that I saw the expected trajectory drawn with dotted line on RViz when I run the Autoware on AWSIM.
However, when I run Autoware on Carla with the bridge, I cannot see any of the dotted lines in RViz as below:

[image]

I saw that the ego vehicle stops when the NPC Vehicle or other obstacles are in front of him, but in the case of the video below, the ego vehicle does not stop and eventually collide to the tail of the NPC Vehicle.

https://github.com/evshary/autoware_carla_launch/assets/50267797/476c7c5b-44a9-4de2-84b5-26d5dba34f56

Could you please check whether the expected trajectory shows well on your screen?
Thanks in advance! I always appreciate to your hard work!

evshary, 2023-07-13:
No, I didn't see that before, because I haven't tried to add NPC in my scenario.
Actually, there is still something worth investigating in the bridge.
For example, Autoware can't detect the traffic light in Carla and follow the traffic rules.
These topics are pretty interesting. Maybe I can dive deep into them after tuning the bridge performance and dealing with version issues.

Kim-mins, 2023-07-13:
Thank you for answering me! I think I should test the bridge more too.

By the way, aren't the traffic light and traffic rules(especially speed limit and stop line) encoded in the map?
I read .osm file of `nishishinjuku` map and saw the code as below:

- traffic light
```xml
<way id="1062" version='1'>
  <nd ref="122948"/>
  <nd ref="122949"/>
  <tag k="type" v="traffic_light"/>
  <tag k="subtype" v="red_yellow_green"/>
  <tag k="height" v="0.450000"/>
</way>
```

- stop line
```xml
<way id="1078" version='1'>
  <nd ref="15115"/>
  <nd ref="9285"/>
  <tag k="type" v="stop_line"/>
</way>
```

- speed limit
```xml
<relation id="4" version='1'>
  <member type="way" role="left" ref="361"/>
  <member type="way" role="right" ref="362"/>
  <member type="relation" role="regulatory_element" ref="1292"/>
  <tag k="type" v="lanelet"/>
  <tag k="subtype" v="road"/>
  <tag k="speed_limit" v="50"/>
  <tag k="location" v="urban"/>
  <tag k="one_way" v="yes"/>
  <tag k="participant:vehicle" v="yes"/>
</relation>
```

I do not know whether Autoware perceives stop lines, but I saw the traffic light detection module is enabled when I run Autoware, so maybe he can abide by the rules if the map provides enough information.. I think.

closed 2023-07-19 by Kim-mins