evshary/autoware_carla_launch #16: Autopilot NPC vehicles often slide through the road

Kim-mins, 2023-07-12:
Hi evshary!

I got a question regarding the behavior of NPC Vehicles.
When I run the simulation with autopilot NPC vehicles, the vehicles often slide through the road randomly.

In [video1](https://drive.google.com/file/d/1HoS05QZLkmBaw9hjaNyeJNs_h7QGxgsd/view?usp=sharing), when the ego vehicle turns to the right, the NPC vehicle is already slided too much, and seems like he invaded the lane while cornering.
Also, you can see that another NPC vehicle invades the lane where the ego vehicle is, and eventually makes a collision with the ego vehicle.

In [video2](https://drive.google.com/file/d/1o5-OfghEU34SshtZ1iPAQhHqLMNb7_0B/view?usp=sharing), the NPC vehicle on the opposite lane stops on the red light sign, but slides through when he starts to move, and eventually collides with the ego vehicle.

Maybe these slidings could be from the weather I configured(especially puddle, rain or wetness), but I don't know why NPC vehicles drive well and abruptly lose control.
Also I remember that I didn't experience this issue while running a traffic simulation with autopilot NPC vehicle on Carla independently(which means w/o bridge and Autoware).

Have you ever experienced the situation like this? or do you know any possible solution on this issue?

evshary, 2023-07-13:
I didn't try running autopilot NPC in the scenario before.

There are several possible reasons:
1. The bridge affects the performance of NPC vehicles while interacting with Carla
2. Carla has some issues while running too many NPC vehicles
3. Maybe there are some issues with autopilot in Carla https://github.com/carla-simulator/carla/discussions/4781

I'm not sure which one is the root cause. But if you can run many autopilot NPC vehicles successfully without bridge, then it might be caused by the first one.

Kim-mins, 2023-07-13:
Thank you for listing the possible reasons!

I also searched for some reasons why they slide through the road, but it seems like the cause is from Carla.
[The issue here](https://github.com/carla-simulator/carla/issues/5693) saids that a friction of one vehicle is applied to all the vehicles in the simulator world.
I did not tell you that I'm changing the friction, so sorry for confusing you, but when I removed the code changing the friction, the NPC vehicles looks work well.

closed 2023-07-13 by Kim-mins