evshary/autoware_carla_launch #23: humble branch is Failed to enter Docker

Autostone-c, 2023-08-03:
[image]

I follow the README.md. and I have already apt install tmux inetutils-ping vim.

evshary, 2023-08-03:
I believe the issue came from the docker cache.
Please do `docker system prune` and try again.

BTW, I saw you tried to use galactic with containers. There is some issue with galactic in container.
Suggest using humbe in the container instead, or running galactic in your native host.

Autostone-c, 2023-08-03:
> I believe the issue came from the docker cache. Please do `docker system prune` and try again.
> 
> BTW, I saw you tried to use galactic with containers. There is some issue with galactic in container. Suggest using humbe in the container instead, or running galactic in your native host.

i will try it. by the way, i follow your README.md. and i see the run-bridge-docker.sh(humble branch) use galactic
[image]
[image]

evshary, 2023-08-03:
Oh, I see. Then that's fine though, since the bridge still depends on some galactic ROS packages.
If Autoware is running on humble, then it's OK.

Autostone-c, 2023-08-03:
> Oh, I see. Then that's fine though, since the bridge still depends on some galactic ROS packages. If Autoware is running on humble, then it's OK.

have you ever try run this work on two machine？for example (carla on x86 workstation and autoware-humble on Nvidia Jetson Orin)

Autostone-c, 2023-08-03:
hi, I have encountered the following error, can you help me
[image]

evshary, 2023-08-04:
> > Oh, I see. Then that's fine though, since the bridge still depends on some galactic ROS packages. If Autoware is running on humble, then it's OK.
> 
> have you ever try run this work on two machine？for example (carla on x86 workstation and autoware-humble on Nvidia Jetson Orin)

It definitely can run on two machines. This is one of the purposes of the projects.
However, I haven't tried Autoware Humble in Jetson Orin and don't know whether there is any issue.

evshary, 2023-08-04:
> hi, I have encountered the following error, can you help me [image]

Maybe you can make sure [Autoware in docker](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-prebuilt/) can run in your environment first.
It seems like there are some issues for your container to use GPU.

Autostone-c, 2023-08-04:
> Maybe you can make sure [Autoware in docker](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-prebuilt/) can run in your environment first. [...]

im trying it? can i have you wechat for communication

evshary, 2023-08-04:
I prefer to have the discussion here because the issue might also be met by others.
Then they could refer to the discussion.

Autostone-c, 2023-08-04:
> I prefer to have the discussion here because the issue might also be met by others. Then they could refer to the discussion.

i see, im trying build autoware in docker. and my ultimate goal is Orin(autoware-humble) and carla on x86 workstation

evshary, 2023-08-04:
> i see, im trying build autoware in docker. and my ultimate goal is Orin(autoware-humble) and carla on x86 workstation

I believe this is feasible, but I have little experience running Autoware in Orin.
Maybe you can take a look at here
https://github.com/orgs/autowarefoundation/discussions/3537

Autostone-c, 2023-08-07:
> I prefer to have the discussion here because the issue might also be met by others. Then they could refer to the discussion.

hi，i can run it on the same machine now, but The Carla scene remains unchanged with the changes in autoware（no car ， and always like the picture below）
[image]

Kim-mins, 2023-08-07:
@Autostone-c 
If you want to follow the driving of the ego vehicle, please provide the option `--pygame` for the [running script](https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-bridge.sh)

Autostone-c, 2023-08-07:
> @Autostone-c If you want to follow the driving of the ego vehicle, please provide the option `--pygame` for the [running script](https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-bridge.sh)

nice，it really help me. thanks

Autostone-c, 2023-08-07:
now, im trying run it on two machine(carla on x86 workstation and autoware-humble on Nvidia Jetson Orin)

Expect good results

Autostone-c, 2023-08-09:
@evshary hi, i try run it on two machine(carla on x86 workstation and autoware-humble on Nvidia Jetson Orin). and i modify the autoware image amd64 to arm64. Nvidia docker run other container has no error.
[image]

evshary, 2023-08-09:
Do you install NVIDIA Container Runtime? I'm not familiar with this part, but I'd suggest checking whether you can run containers with NVIDIA GPU first.

Autostone-c, 2023-08-09:
> Do you install NVIDIA Container Runtime? I'm not familiar with this part, but I'd suggest checking whether you can run containers with NVIDIA GPU first.

yes， i install it. now i delete --nvidia  ,and enter into docker. i want ask if i run two machine, should the configuration file be modified? like ip or some else

closed 2023-08-09 by Autostone-c