evshary/autoware_carla_launch #24: Mission planner cannot be started for second vehicle

zeidk, 2023-08-03:
Hello,
I am using zenoh with two vehicles.
It seems that mission planner only works for the first vehicle spawned in autoware.
Steps to reproduce:

- 1:
```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <group>
    <executable cmd="$(env AUTOWARE_CARLA_ROOT)/run_carla_bridge.sh" output="screen">
      <env name="RUST_LOG" value="z=info"/>
    </executable>
    <executable cmd="poetry run python3 main.py --host $(env CARLA_SIMULATOR_IP) --rolename 'rv1' --position 396.0,250.0,0.300000,0.000000,-90.000053,0.000000" cwd="$(env AUTOWARE_CARLA_ROOT)/external/zenoh_carla_bridge/carla_agent" output="screen" />
    <executable cmd="poetry run python3 main.py --host $(env CARLA_SIMULATOR_IP) --pygame --rolename 'rv2' --position 396.0,316.0,0.300000,0.000000,-90.000053,0.000000" cwd="$(env AUTOWARE_CARLA_ROOT)/external/zenoh_carla_bridge/carla_agent" output="screen" />
  </group>
</launch>
```

- 2: 
```
ROS_DOMAIN_ID=1 VEHICLE_NAME="rv1" ros2 launch autoware_carla_launch autoware_zenoh.launch.xml
```

- 3 :
 ```
ROS_DOMAIN_ID=2 VEHICLE_NAME="rv2" ros2 launch autoware_carla_launch autoware_zenoh.launch.xml
```

- 4: In RViz, when I set the goal for rv1 then a path is computed. When I set a goal for rv2 then a plan is not computed. If I start step 3 before step 2 then I can only set the goal for rv2 and not rv1.

Any idea what is happening?

Thank you

evshary, 2023-08-04:
Unfortunately, I cannot reproduce your issue. I can set goals for both of my two RViz.
Are you using the latest commit in the galactic branch? Also, does it work fine in the past, and have problems recently?

zeidk, 2023-08-07:
I will give it another try using the original launch files that come with your repository. I have modified the original launch files and I probably messed up a few things.

evshary, 2023-12-25:
I'll close the issue since it has been a while.
Feel free to reopen it if you have any further questions.

closed 2023-12-25 by evshary