evshary/autoware_carla_launch #25: ros2 cli or dev tools with zenoh bridge

harishkumarb-tw, 2023-08-04:
I am new to zenoh dds bridge. When my autoware is running, I couldn't see topics or nodes using ros2 cli. I get empty output when I run `ros2 node list` or `ros2 topic list`. 
I did exec into the humble docker container and ran `ros2 topic list` while the autoware is running in another terminal(same container),
[image]
What am I missing? How to use all the ros2 dev tools(cli, rqt, rosbag, etc) using the zenoh bridge(for carla data)?

evshary, 2023-08-04:
Do you run `source env.sh` before `ros2 topic list`?
Since there are some environment settings about ROS in `env.sh`, I guess it'll affect the ros2 cli tools
https://github.com/evshary/autoware_carla_launch/blob/humble/env.sh#L16

harishkumarb-tw, 2023-08-04:
Thanks for the quick reply. It works now after sourcing 👍. 
But, is there anyway to bridge the carla data/topics alone without running the whole `autoware.launch`?

evshary, 2023-08-04:
zenoh-bridge-dds has two modes: mirror and forward. I use mirror mode here, which is the default.
Only forward mode supports commands like `ros2 topic list`, but the performance might be worse.
I don't use the mode because I've already known all the topics I need, but maybe it is worth giving it a try.
For more detail, you can check here.
https://github.com/eclipse-zenoh/zenoh-plugin-dds#full-support-of-ros-graph-and-topic-lists-via-the-forward-discovery-mode

Another way is to create a dummy ROS node that subscribes to the topic you need, then zenoh-bridge-dds would be able to bridge the data/topics for them.

datvuthanh, 2023-10-30:
Hi @harishkumarb-tw , i ran this script source env.sh but it didn't work. Could you give me some suggestions?

evshary, 2023-10-31:
> Hi @harishkumarb-tw , i ran this script source env.sh but it didn't work. Could you give me some suggestions?

Do you mean `ros2 topic list` not work? If so, try `ros2 daemon stop` and then try again. (remember to source env.sh)
It would be great if you provide more detailed information.

datvuthanh, 2023-10-31:
hi @evshary , this didnt work for me. 
[image]

evshary, 2023-11-01:
@datvuthanh You need to run `ros2 topic list` in the same container.
I think you can use terminal multiplexer like tmux to run Autoware and ros2 command at the same time.

datvuthanh, 2023-11-01:
@evshary thank you, it works now.

evshary, 2023-12-25:
Close the issue while it's completed

closed 2023-12-25 by evshary