evshary/autoware_carla_launch #26: two machine configuration files

Autostone-c, 2023-08-09:
if run this work on two machines. any configuration files should be modify?

evshary, 2023-08-10:
Are you running in the same LAN?
If so, I believe you don't need to modify anything.
If not, you can check the script and assign the IP of Carla host
https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-autoware.sh
Something like `./run-autoware.sh v1 <Carla's IP>`

Autostone-c, 2023-08-10:
> Are you running in the same LAN? If so, I believe you don't need to modify anything. If not, you can check the script and assign the IP of Carla host https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-autoware.sh Something like `./run-autoware.sh v1 <Carla's IP>`

yes, i am running in the same LAN. Only WiFi was used, but no switches were used。I don't know if it's because autoware docker is on Orin, and this job doesn't work for me

Autostone-c, 2023-08-10:
> > Are you running in the same LAN? If so, I believe you don't need to modify anything. If not, you can check the script and assign the IP of Carla host https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-autoware.sh Something like `./run-autoware.sh v1 <Carla's IP>`
> 
> yes, i am running in the same LAN. Only WiFi was used, but no switches were used。I don't know if it's because autoware docker is on Orin, and i modify the Dockerfile_autoware "FROM ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64" to "FROM ghcr.nju.edu.cn/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-arm64".Is this place causing communication failure？

i add ros2/demos.git to autoware_carla.repos and rebuild autoware on Orin. then x86 (bridge docker) run talker, Orin(autoware docker) run listener. it can communicate

Autostone-c, 2023-08-10:
> i add ros2/demos.git to autoware_carla.repos and rebuild autoware on Orin. then x86 (bridge docker) run talker, Orin(autoware docker) run listener. it can communicate

The current status is that the rviz map is black and always shows “waiting init pose”

evshary, 2023-08-10:
Hmm, I guess this means the Zenoh does not connect to each other.
Could you give a try to add IP of Carla?
Like this `./run-autoware.sh v1 <Carla's IP>`

Also, if you can attach the log, it will help to solve the issue.

Autostone-c, 2023-08-10:
> 
[launch.log](https://github.com/evshary/autoware_carla_launch/files/12311346/launch.log)

i modify some parameters as below。 because i cant enter into docker use "--nvidia"
[image]

evshary, 2023-08-11:
Could you provide the log under `bridge_log` folder?
I think Autoware didn't receive messages from bridge based on your launch.log.

Autostone-c, 2023-08-11:
> Could you provide the log under `bridge_log` folder? I think Autoware didn't receive messages from bridge based on your launch.log.

ok, which machine bridge_log should i give you? Orin(autoware-docker) or x86 station(bridge-docker)?

evshary, 2023-08-11:
Log from bridge-docker.
BTW, Have you tried to add IP for `./run-autoware.sh v1 <Carla's IP>`?
Is the result still the same?

Autostone-c, 2023-08-11:
> Log from bridge-docker. BTW, Have you tried to add IP for `./run-autoware.sh v1 <Carla's IP>`? Is the result still the same?
[bridge_log.zip](https://github.com/evshary/autoware_carla_launch/files/12318071/bridge_log.zip)
yes, i had try ./run-autoware.sh v1 <my_Carla's IP> , same as default.

evshary, 2023-08-11:
It's a little weird that Zenoh is only binding to 127.0.0.1...

```
 INFO  zenoh_carla_bridge > Running Carla Autoware Zenoh bridge...
 INFO  zenoh::net::runtime > Using PID: 42f1ee47ee7735f8c7f9afe0cb4e3ef
 INFO  zenoh::net::runtime::orchestrator > Zenoh can be reached at: tcp/127.0.0.1:7447
 INFO  zenoh::net::runtime::orchestrator > zenohd listening scout messages on 224.0.0.224:7446
```

Can you check the network settings in `bridge-docker` container?

Autostone-c, 2023-08-11:
> It's a little weird that Zenoh is only binding to 127.0.0.1...
>
> ```
>  INFO  zenoh_carla_bridge > Running Carla Autoware Zenoh bridge...
>  INFO  zenoh::net::runtime > Using PID: [...]

what should i do? can you share some bash order to me（like: ip a）.` I am completely clueless about network settings

evshary, 2023-08-11:
1. List IP on all interfaces
2. Check whether the port 7447 is binded by Zenoh in carla-bridge container
3. Can carla-bridge and autoware container ping each other directly?

If there are still no issues, then might need to check whether Zenoh example can communicate with each other in two containers.

These items just come to my mind now.

evshary, 2023-08-16:
@Autostone-c I guess I found the root cause. You can try the latest humble branch.
I fixed the issue here.
https://github.com/evshary/autoware_carla_launch/pull/28

Autostone-c, 2023-08-16:
> @Autostone-c I guess I found the root cause. You can try the latest humble branch. I fixed the issue here. #28

i will try. and i think perhaps i need another x86 workstation to verify this work. maybe the error due to Orin

evshary, 2023-08-16:
> i will try. and i think perhaps i need another x86 workstation to verify this work. maybe the error due to Orin

You can try the fix first. I think this can solve the problem above, although I'm not sure whether there is any other issue running on Orin. :sweat_smile:

Autostone-c, 2023-08-16:
> > i will try. and i think perhaps i need another x86 workstation to verify this work. maybe the error due to Orin
> 
> You can try the fix first. I think this can solve the problem above, although I'm not sure whether there is any other issue running on Orin. 😅

hi, i have done it. and i can see all topic run ros2 topic list. but the /map/vector_map only has one frame data. and initalpose has no data. i doubt i delete --nvidia in run-autoware-docker.sh causing some issues
[image]

evshary, 2023-08-17:
> hi, i have done it. and i can see all topic run ros2 topic list. but the /map/vector_map only has one frame data. and initalpose has no data. i doubt i delete --nvidia in run-autoware-docker.sh causing some issues [image]

Hmm.. I guess so, because without `--nividia` means you don't use GPU in your container.

Autostone-c, 2023-08-17:
> Hmm.. I guess so, because without `--nividia` means you don't use GPU in your container.

but i saw use "nvidia-container-runtime" at /etc/docker/daemon.json。It's been bothering me for a long time

evshary, 2023-08-24:
@Autostone-c Just FYI
https://github.com/orgs/autowarefoundation/discussions/3764

Autostone-c, 2023-08-30:
> @Autostone-c Just FYI https://github.com/orgs/autowarefoundation/discussions/3764

hi， i have done it.  This job is useful to me。sincere thanks

closed 2023-08-30 by Autostone-c