evshary/autoware_carla_launch #32: Zenoh Bridge Limitations

MarioAlmelaUMH, 2023-09-18:
Hello, I am currently working with Zatitech's Openplanner bridge (https://github.com/ZATiTech/open_planner). I've heard the Zenoh bridge is based on Zatitech's, and I was wondering if it has inherited the latter's limitations. 

Currently, Openplanner bridge only supports one sensor each (one camera, one lidar, one GNSS and one IMU). Can you modify this sensor kit in Zenoh's bridge (for example, adding multiple cameras or lidars)? 

Additionally, I've had problems when assigning each sensor a different _sensor_tick_ parameter (in order to change the frequency of each sensor), so right now the system only works with all sensors working at the same frequency. Is it possible to assign different sensor frequencies on Zenoh's bridge?

Thank you for your answers.

evshary, 2023-09-19:
Hi

> Hello, I am currently working with Zatitech's Openplanner bridge (https://github.com/ZATiTech/open_planner). I've heard the Zenoh bridge is based on Zatitech's, and I was wondering if it has inherited the latter's limitations.

in fact, Zenoh bridge is not based on Zatitech's. We rewrote the bridge, but using the map generated by Zatitech.

> Currently, Openplanner bridge only supports one sensor each (one camera, one lidar, one GNSS and one IMU). Can you modify this sensor kit in Zenoh's bridge (for example, adding multiple cameras or lidars)?

It's possible to support multiple sensors. In Carla, your sensor has its own name, and the bridge will publish the ROS topic based on that name. For example, the bridge will publish camera's image with the topic "sensing/camera/{sensor_name}/image_raw".
But I haven't tested multiple cameras on the same vehicle yet. If you have any problems, please let me know.

> Additionally, I've had problems when assigning each sensor a different sensor_tick parameter (in order to change the frequency of each sensor), so right now the system only works with all sensors working at the same frequency. Is it possible to assign different sensor frequencies on Zenoh's bridge?

The frequency should be based on Carla setting, so theoretically each sensor should have its own frequency. However, I found that the frequency might not be as fast as you set. For example, if you set 30 FPS on camera, the bridge might only publish images at 15 Hz while there are several sensors. I'm not really sure if it's the performance limitation on Carla or if there is anything wrong with the bridge. Need more investigation.

MarioAlmelaUMH, 2023-10-02:
Hello, does this bridge support different sync modes? Currently I'm working with Carla's Synchronous mode, is Asynchronous also supported?

Thank you.

evshary, 2023-10-03:
> Hello, does this bridge support different sync modes? Currently I'm working with Carla's Synchronous mode, is Asynchronous also supported?
> 
> Thank you.

No, the bridge only supports synchronous mode now.

MarioAlmelaUMH, 2023-10-03:
Thank you for your answer. I was wondering, is it possible to change the Carla map that the bridge uses? Currently it automatically sets Town01, is there a config option to change this, add more vehicles (Non-ego, Carla controlled), change the weather, etc.?

evshary, 2023-10-04:
> I was wondering, is it possible to change the Carla map that the bridge uses? Currently it automatically sets Town01, is there a config option to change this, add more vehicles (Non-ego, Carla controlled), change the ego vehicle's initial position, etc.?

To change the map, please refer to the link
https://github.com/evshary/autoware_carla_launch/tree/humble/carla_map
For Carla settings, you can revise the Python script here
https://github.com/evshary/zenoh_carla_bridge/tree/4413e2907c807826c4e3f3df1e4aec74a5353774/carla_agent
To change the initial pose, you can add the argument to the Python script
https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-bridge-two-vehicles.sh#L13

> Also, I've tried to run ros2 topic list inside the docker container, but only topics /parameter_events and /rosout show up. Is there a way to see Autoware topics?

Remember to run `source env.sh` before ros2 command. However, since the bridge sends Zenoh data to zenoh-bridge-dds, you can't run `ros2 topic list` to get the messages from the bridge unless you've already run Autoware in your container. I would suggest using tmux or other terminal multiplexers in the container. Then you can run Autoware in one terminal and run the ros2 command in another one to see the detail.

MarioAlmelaUMH, 2023-10-04:
Thank you so much. I've followed this steps and now I'm able to see the complete list of topics. I still have a couple issues, however.
- When I try to echo a topic, it throws an error as the Autoware specific message type cannot be determined. I have tried to source the environment with source env.sh, but it doesn't seem to work when using tmux (this can be seen in the right panel in the image below). I already sourced the environment on the left panel before starting the tmux session:

[image]

I need to echo the topics so I can check if the modules are working properly.

- On another topic, I may want to change the ego vehicle's sensor kit, to include multiple cameras and lidars. I suppose there will be 2 config files, one for Carla and one for Autoware, and both need to have the same sensor config. Is this right? Which files should I modify?

Thank you so much for your answers.

evshary, 2023-10-05:
For the first one, this is because in tmux `$0` is `-bash` not `bash`. For the temporary workaround, please run bash after splitting a new terminal in tmux. I'll see how to fix it.

About the second topic, you're right. You need to add extra sensors for the vehicle in Carla by yourselves
https://github.com/evshary/zenoh_carla_bridge/blob/main/carla_agent/simulation/world.py#L118
For the Autoware, you can refer to the official documentation
https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-description/creating-vehicle-and-sensor-description/
The sensor_kit_description and sensor_kit_launch are here
https://github.com/evshary/autoware_carla_launch/tree/humble/src

MarioAlmelaUMH, 2023-10-05:
Thank you. I've been able to echo some topics and source the environment properly. I'm trying to record a ros2bag using this command:
`ros2 bag record /sensing/camera/traffic_light/image_raw /sensing/imu/tamagawa/imu_raw /sensing/lidar/concatenated/pointcloud /sensing/gnss/pose_with_covariance /perception/object_recognition/detection/clustering/clusters /perception/object_recognition/detection/clustering/objects /perception/object_recognition/detection/objects /perception/object_recognition/objects /system/component_state_monitor/component/autonomous/perception /control/command/control_cmd /perception/object_recognition/detection/rois0 perception/object_recognition/detection/camera_lidar_fusion/objects /clock -o <bag_name> --qos-profile-overrides-path qos_override.yaml`

However, when I do this, the following error appears:
[image]

The ros2bag isn't even subscribing to this topic, yet it seems it has two types associated to it. I have never encountered this error before, so I have no clue as to what may be happening.

Thank you for your answers.

evshary, 2023-10-06:
Thanks for your report.
I can reproduce the issue but might need further investigation.

evshary, 2023-10-13:
For the second one, I publish the camera_info with this name `/sensing/camera/traffic_light/camera_info` by default
However, if there is no any subscriber, then you can't see it.
Maybe you can check which topic name `camera_lidar_fusion` is subscribing.

MarioAlmelaUMH, 2023-10-13:
Thank you for your answers. Regarding the second question, I've checked the file `camera.launch.xml` and I've seen the output topic of the camera as `sensing/camera/traffic_light/camera_info`. 

The `camera_lidar_fusion` node listens to topics of the form `sensing/camera/camera*/camera_info`, with (*) being the number of the camera. I changed the topic in `camera.launch.xml` manually to `sensing/camera/camera0/camera_info`, but it doesn't seem to work. There are deeper Autoware files like `perception.launch.xml`, which define in more depth this camera topics. However, I can't find them in my current installment of the bridge. I've tried searching outside and inside the docker container, but I still haven't found these files. Where is Autoware proper located inside the bridge?

Thank you.

evshary, 2023-10-16:
You also need to change the camera name from Python script
https://github.com/evshary/zenoh_carla_bridge/blob/main/carla_agent/simulation/world.py#L124
Then the bridge will use your name to publish the topic.

For the Autoware, it's a little tricky. I found that the current Autoware container doesn't include the source code (Only binary). It's not convenient to do development, but you can try to clone your own Autoware and put it into the container. No need to clone the whole Autoware. You can only get part of the component you want to modify in Autoware and overlay the original one.

MarioAlmelaUMH, 2023-10-16:
Thanks for your answer. I'm looking at the makefile and the dockerfiles of the bridge, and I don't know where exactly should I put the clone statement. In those files, I haven't seen the instruction that introduces the binary Autoware files inside the container (unless they are already inside the image that gets downloaded when you first run the bridge). I'm not very experienced yet with docker and docker images. How could I clone the original Autoware inside the container? I'm guessing I should add the clone command to one of the dockerfiles then rebuild the bridge, but how can I override the current binary files?

Thank you .

evshary, 2023-10-18:
The Autoware binary files are already in the base image (ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64). You can look at `/autoware` folder, which includes the binary files.

To overlay the existing Autoware packages, you can find out which package you want to use first and put it under `autoware_carla_launch/src` folder. After rebuilding the project, you'll use the package cloned by yourself.

You can refer to [here](https://github.com/evshary/autoware_carla_launch/blob/humble/autoware_carla.repos#L6). I overlay the autoware_launch by myself to avoid some issues.

MarioAlmelaUMH, 2023-10-24:
Thank you. I'm trying to overlay the perception module so I can get access to the tensorrt yolo nodes. However, when I try to rebuild the bridge, it throws an error because it's missing cuDNN library:

[image]

I've tried to install both CUDA and cuDNN by modifying the makefile (the prepare_autoware command) and it seems to install the library properly, however the build_autoware phase still throws that error. 

[image]

How should I install cuDNN properly?

Thank you.

datvuthanh, 2023-10-30:
Hi @MarioAlmelaUMH , i run source env.sh but it didn't work. Could you give me some suggestions?

MarioAlmelaUMH, 2023-10-30:
I'm not an expert on the matter either. For me, it has always worked fine. What kind of problem did you have?

evshary, 2023-11-22:
@MarioAlmelaUMH I added some script for building your own Autoware, but now only supports Carla 0.9.14 (Need to port back to Carla 0.9.13)
https://github.com/evshary/autoware_carla_launch/tree/humble-0.9.14#build
If you're still struggling with building Autoware in the docker, maybe you can try this one.

By the way, I found that we would face some CUDA issues while building with the prebuilt image (humble-2023.08-prebuilt-cuda-amd64), so I used another image (humble-2023.08-cuda-amd64) instead.
Also, I use 2023.08 instead of 2023.07 to make the code consistent. The date of the code in tag 2023.07 is 2023.07.29, but the code in prebuilt binary is 2023.07.15. It'll cause some conflicts.

MarioAlmelaUMH, 2023-11-22:
Ok, thank you. I still have to try the first option, but if that gives me trouble, I will try out this new one. I will also take into account the images used. I have to run cuda so better to start with humble-2023.08-cuda-amd64. I will keep updating as I discover new things.

evshary, 2023-12-08:
Hi @MarioAlmelaUMH, I believe the key problem here is that you also need to bump the version of autoware_launch to version carla_202308. Here is the commit I did: https://github.com/evshary/autoware_carla_launch/commit/467b7c91b5d1d1ff5fc8297322dda80d417ef228#diff-66a6d5b656005e10aade7bf7a5c9c7722252f3800d4be3f0e5cf14f4847432cc

I picked the commits for Carla 0.9.13, and pushed them to the branch support_src_for_humble
I haven't tested it properly yet (traveling now), but maybe you can try that and see if it works for you or not.
https://github.com/evshary/autoware_carla_launch/tree/support_src_for_humble#build-the-container-for-zenohautoware
If everything is ok, I'll merge it into the humble branch

MarioAlmelaUMH, 2023-12-11:
Hi @evshary . I have tested both options you showed me, but unfortunately they haven't worked yet.
I'll start with the last option. I visited the github page you showed (https://github.com/evshary/autoware_carla_launch/tree/support_src_for_humble#build-the-container-for-zenohautoware) and followed the instructions of the optional source installation part. I run the script **build-autoware.sh** and the proceeded as told with make prepare_autoware and make_build_autoware. Everything seemed fine, but when I tries to run the bridge as normally, this error popped up:

[image]

I'll include the log files for more detail: 
[zenoh_bridge_dds.log](https://github.com/evshary/autoware_carla_launch/files/13635225/zenoh_bridge_dds.log)
[autoware.log](https://github.com/evshary/autoware_carla_launch/files/13635227/autoware.log)

Basically, I think the `carla_autoware_launch` node is not being found. I have tried to rebuild the bridge a couple times, but I'm not sure if it is building correctly. Also, aparently there's a config file whose path is missing.

----------------------------------------------------------------------

On another note, for the first option, I bumped the autoware_launch version to carla_202308, but this didn't work either. I had to copy and paste multiple folders into Autoware (`carla_pointcloud`, `carla_sensor_kit_description` and `carla_sensor_kit_launch`), as well as modify the `env.sh` file to source Autoware's `setup.bash` instead of the bridge's.

With all, I was able to launch Autoware, but the ROS Time was frozen at 0 and the Autoware dds didn't seem to be connected to the `zenoh_carla_bridge`. After I tried the second option (https://github.com/evshary/autoware_carla_launch/tree/support_src_for_humble#build-the-container-for-zenohautoware) I was not able to rebuild the first container again and reproduce the issue, being stuck in this new error:
[image]

evshary, 2023-12-11:
@MarioAlmelaUMH For the first question, can you check whether you run `source env.sh` or not?
I believe you are missing some variables.
https://github.com/evshary/autoware_carla_launch/blob/support_src_for_humble/env.sh#L13
Remember you still need to run `source env.sh` after building.

For the second question, I believe the package `autoware_carla_launch` is not built. Can you check whether it exists in install folder?

MarioAlmelaUMH, 2023-12-14:
Thanks for your answer, @evshary . I tried again the first option (following this link: https://github.com/evshary/autoware_carla_launch/tree/support_src_for_humble#build-the-container-for-zenohautoware) and this time it worked. We are now exploring Autoware's source code in order to find how can we modify the sensor kit.

Since that last part belongs to this thread (https://github.com/evshary/autoware_carla_launch/issues/43), I think it's safe to finally close this one. I'll reopen it if more questions come afloat, but for now, I think it's ok.

Thank you so much.

closed 2023-12-14 by MarioAlmelaUMH