evshary/autoware_carla_launch #33: Ego vehicles not moving

MarioAlmelaUMH, 2023-10-02:
Hello, I've followed the tutorial shown in: https://autoware.org/running-multiple-autoware-powered-vehicles-in-carla-using-zenoh/. I have successfully installed the system and both Rviz visualizers opened up. However, when I set the Goal Pose, the vehicles couldn't move. Is there any configuration steps I missed? Is it possible for the problem to be related to internet connection? I appreciate any insight you may have.

Thank you so much for your answers.

evshary, 2023-10-03:
You're using Humble version right?
Do you mean the vehicles still don't move after setting the goal and clicking auto?
Maybe you can try only one vehicle first and see whether the problem exists or not.
I'm not really sure if it's a performance issue or not.
If there is any video that can be provided, it might be helpful.

MarioAlmelaUMH, 2023-10-03:
Thank you for your answer.
I am using the humble version indeed.
I tried to set up only one vehicle as suggested, and the vehicle is able to move now. However, the trajectory becomes unstable and the vehicle ends up crashing out of the road. Seems like Autoware has trouble following the vehicle's movement in Carla.

Also, I don't know if this is an issue, but the Auto button is normally greyed out, only activating every now and then for brief amounts of time before deactivating again. By clicking Local and then Stop, sometimes Autonomous mode just activates by itself.

I have recorded a video that shows the ego vehicle falling out of the road. This, as for now, has happened always, the vehicle has never properly arrived to the goal.

https://drive.google.com/file/d/1oLYfMF9eWgneoosvLbdFm1GDhT35TyUo/view?usp=sharing

evshary, 2023-10-04:
> Thank you for your answer. I am using the humble version indeed. I tried to set up only one vehicle as suggested, and the vehicle is able to move now. However, the trajectory becomes unstable and the vehicle ends up crashing out of the road. Seems like Autoware has trouble following the vehicle's movement in Carla.
> 
> Also, I don't know if this is an issue, but the Auto button is normally greyed out, only activating every now and then for brief amounts of time before deactivating again. By clicking Local and then Stop, sometimes Autonomous mode just activates by itself.
> 
> I have recorded a video that shows the ego vehicle falling out of the road. This, as for now, has happened always, the vehicle has never properly arrived to the goal.
> 
> https://drive.google.com/file/d/1oLYfMF9eWgneoosvLbdFm1GDhT35TyUo/view?usp=sharing
> 
> As you can see in the video, ROS Time seems a bit inconsistent. I've seen in Carla documentation (https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/) that using synchronous mode without a fixed time-step may cause problems with the physics. Could this be related to the ego vehicle falling out of the road?

You're right. This is related to the ROS Time. I've found this issue before, but then disappear and can't be reproduced after some tries. I haven't found a way to reproduce this issue. I guess this is because the publisher of ROS time has an unstable frequency. I'll take a look at that.

evshary, 2023-10-06:
@MarioAlmelaUMH  Can you give the branch a try? I've done some refactors, but I can't reproduce your issue, so not sure whether the issue is gone or not.
https://github.com/evshary/autoware_carla_launch/tree/update_bridge

MarioAlmelaUMH, 2023-10-06:
I have installed the branch and made the ego vehicle run through maps Town01 and Town02, with and without Carla controlled vehicles. So far, I haven't encountered the issue anymore. The ego vehicle follows the road correctly. Thank you so much for your help.

I'm going to close this issue for now, and I'll reopen it if the problem appears again.

closed 2023-10-06 by MarioAlmelaUMH