evshary/autoware_carla_launch #38: Error in make prepare_bridge

VietPT3502, 2023-10-28:
I got this error when running make prepare_bridge
# Get code
git submodule update --init --recursive
fatal: detected dubious ownership in repository at '/root/autoware_carla_launch'
To add an exception for this directory, call:

	git config --global --add safe.directory /root/autoware_carla_launch
make: *** [Makefile:16: prepare_bridge] Error 128

if im using git config --global --add safe.directory, i will have the same error with [issue 3](https://github.com/evshary/autoware_carla_launch/issues/3)

VietPT3502, 2023-10-28:
fixed by running sudo chown -R root:root ~/autoware_carla_launch/ in docker

closed 2023-10-30 by VietPT3502

evshary, 2023-10-30:
Out of curiosity, do you use sudo to run the container? I think the user name should not be root.

VietPT3502, 2023-10-30:
yes im using sudo ./container/run-bridge-docker.sh