evshary/autoware_carla_launch #39: New fresh installment of Zenoh bridge can't communicate between hosts

MarioAlmelaUMH, 2023-10-30:
Hello, I have a modified version of Zenoh bridge, where I'm trying to embed source Autoware into the container. However, I need to run some tests with vanilla bridge and I installed a fresh version on my computer alongside the modified one.

After building the bridge, I've noticed that the carla half of it (run-bridge.sh) and the autoware half (run-autoware.sh) have problems communicating. The ROS Time isn't moving at all and the simulation is frozen:

[image]

I've tried checking the IP of the computers I'm using, delete and re-download the images, and clearing docker's cache with `docker builder prune`. However, nothing's changed. Has anyone have any insight as to why this is happening? I've noticed some environment variables can affect other installations of the bridge, can this be related? 

I am honestly shocked before the bridge had never failed in its vanilla state. Are there problems with multiple installations of the bridge on the same computer?

Thank you so much for your answers.

evshary, 2023-10-31:
Could you share the log with me? I rebuilt the autoware_carla_launch again, but it works well.
By the way, I've updated autoware_carla_launch recently. Are you using the latest one or the one you cloned before?

MarioAlmelaUMH, 2023-10-31:
Here, I've zip-filed the log directory.

[build_2023-10-30_12-27-59.zip](https://github.com/evshary/autoware_carla_launch/files/13212972/build_2023-10-30_12-27-59.zip)

I've used the `git clone https://github.com/evshary/autoware_carla_launch.git -b humble` for installing the bridge. I supposed it contained the latest changes, but maybe I'm wrong.

evshary, 2023-10-31:
oh, I mean the log under `bridge_log` folder and also `~/.ros/log` in the container. These are the runtime log while log folder only contains build log.

> I've used the git clone https://github.com/evshary/autoware_carla_launch.git -b humble for installing the bridge. I supposed it contained the latest changes, but maybe I'm wrong.

I believe it contained the latest changes already.

MarioAlmelaUMH, 2023-10-31:
Sorry, wrong computer. 
[bridge.log](https://github.com/evshary/autoware_carla_launch/files/13213169/bridge.log)
[vehicle.log](https://github.com/evshary/autoware_carla_launch/files/13213170/vehicle.log)

evshary, 2023-10-31:
I didn't see anything wrong in the log. I guess there is something wrong with the Zenoh connection between bridge container and autoware container.
Could you pull the latest code again, and share the log under `autoware_log` folder? (I've just updated it)
I have updated the Dockerfile, so you might need to delete the old docker image.

MarioAlmelaUMH, 2023-10-31:
I can't find the `autoware_log` folder. I've searched for it both inside and outside of the container. Also, the `make build_autoware` command gives this error:

[image]

evshary, 2023-11-01:
You need to get the latest code to support putting log into `autoware_log`. This is the feature I added yesterday.

> Also, the make build_autoware command gives this error:

I guess you forgot to `source env.sh` or maybe forgot to `make prepare_autoware` to download Rust.

MarioAlmelaUMH, 2023-11-02:
Thank you. I was able to build the latest version and access the `autoware_log` folder. Here are the logs generated:

[autoware.log](https://github.com/evshary/autoware_carla_launch/files/13235840/autoware.log)
[zenoh_bridge_dds.log](https://github.com/evshary/autoware_carla_launch/files/13235842/zenoh_bridge_dds.log)

evshary, 2023-11-02:
Unfortunately, I still couldn't find anything suspicious in the log.
I guess I need more detailed log to check the connection.
Could you add `RUST_LOG=z=debug` in `run-autoware.sh` and share the log again, please?
[image]

BTW, when you ran `run-autoware.sh`, did you add any other argument, or just `./run-autoware.sh`?

MarioAlmelaUMH, 2023-11-02:
Ok, I added that line and generated these logs:

[autoware.log](https://github.com/evshary/autoware_carla_launch/files/13238687/autoware.log)
[zenoh_bridge_dds.log](https://github.com/evshary/autoware_carla_launch/files/13238688/zenoh_bridge_dds.log)

And yes, I'm adding 2 arguments to `run_autoware.sh`: the vehicle name and the other host's IP (as seen in https://autoware.org/running-multiple-autoware-powered-vehicles-in-carla-using-zenoh/)

evshary, 2023-11-03:
@MarioAlmelaUMH I guess I found the root cause.
Since I change the argument type from IP to IP:port, your IP won't be used.
I fixed it in #41 to use default port while not assigning port.

However, I'm curious about why multicast didn't work in your case. Although the IP:port is wrong, Zenoh can still use scouting to find other peers. Would you mind sharing your `ifconfig` in your native host?
I want to check whether the docker network interface enable MULTICAST or not.

MarioAlmelaUMH, 2023-11-03:
Ok. I have 2 hosts. One of them (the one that holds Autoware) has this `ifconfig`:

[image]

The other has this:

[image]

I think both of them support multicast.

evshary, 2023-11-03:
Then it makes sense if you use two hosts. Docker network interface can't detect other host's container.
Let's see whether the latest update fixes your problem.

> BTW, given that Carla simulator uses port 2000, where should I configure the port for the bridge?

Please add `--carla-port 2000` here. The default is port 2000 already.
https://github.com/evshary/autoware_carla_launch/blob/humble/script/run-bridge.sh#L10

evshary, 2023-11-03:
Maybe let me clarify a bit more, after updating the latest code, you don't need to change anything to run Zenoh Carla bridge successfully.
Also, Carla's port and Zenoh's port are different things. What I've changed is the type of Zenoh connection, so it does nothing to Carla.

MarioAlmelaUMH, 2023-11-03:
Ok. I added port 7447 to the IP when running `run_autoware.sh` and it worked for now. I've been able to use the bridge as usual. I guess if I update the bridge now, it won't be necessary anymore?

evshary, 2023-11-03:
Only IP will be enough. If you don't assign, it'll use 7447 by default.

MarioAlmelaUMH, 2023-11-03:
Ok, thank you. With this answer, this specific thread should be closed. I'll reopen it if I encounter more problems.

closed 2023-11-03 by MarioAlmelaUMH