evshary/autoware_carla_launch #45: Can i use rospy pub and sub topic from carla?

trand2k, 2023-11-13:
Hi author, 
Can i use python API read data from carla and rospy for pub and sub to autoware without zenod ? 
Rust and zenod is big challenge with me ? Have any problem if using only python for connect carla with autoware ? 
Thank you.

evshary, 2023-11-14:
Of course, you can use Python to develop your own bridge to connect Python & Carla.

In my implementation, the main reason I use Zenoh is that Zenoh can
1. Zenoh can distinguish different vehicles' traffic. It's useful to have multiple vehicles simulation.
2. Zenoh can limit the ROS / DDS traffic inside the host and only pass through necessary information.
3. Zenoh can cross LAN. That is, your simulator and Autoware don't need to be in the same LAN.
If the use cases above do not match your scenario, then you don't need to use Zenoh.

Speaking of why use Rust, this is my preference, and also this is the native implementation of Zenoh.

trand2k, 2023-11-14:
> Of course, you can use Python to develop your own bridge to connect Python & Carla.
>
> In my implementation, the main reason I use Zenoh is that Zenoh can
>
> 1. Zenoh can distinguish different vehicles' [...]

Thank for your reply, Do you have any tip for me when build bridge connect Autoware and Carla using python?

evshary, 2023-11-17:
Maybe you can check this Python implementation
However, you might need some modifications to the project. As far as I know, it could only work on galactic.
https://github.com/orgs/autowarefoundation/discussions/2667

evshary, 2023-12-25:
Close the issue since there is no further question.
Feel free to reopen it again if there are any related questions.

closed 2023-12-25 by evshary