evshary/autoware_carla_launch #47: [Question] Control command mapping from Autoware to Carla

Kim-mins, 2023-12-08:
Hi @evshary

I'm keep using your bridge implementation for Autoware and Carla, and I have one question today.
I wonder how the message from `/control/command/control_cmd` to be converted to the control values(throttle, brake, ...) for Carla.
I think the most related part of the code is here:

I'm currently just an end-user of your

evshary, 2023-12-08:
Hi @Kim-mins
Every steps in Carla will call the step function in vehicle bridge, which includes `update_carla_control`
I believe your question is how `update_carla_control` gets the ROS 2 message.
You can take a look at here
https://github.com/evshary/zenoh_carla_bridge/blob/36ff278dc508697a5ef9800b5d32ee5f308e33c9/src/bridge/vehicle_bridge.rs#L134
After the subscriber receives the Ackermann message, it'll store it in `current_ackermann_cmd`, which is used by `update_carla_control`.

After getting the values, we can have some calculations to convert them into throttle, brake...
Also worth mentioning, since Carla 0.9.14 provides API to do the transformation for us, now we don't need to calculate it by ourselves.
https://github.com/evshary/zenoh_carla_bridge/blob/carla-0.9.14/src/bridge/vehicle_bridge.rs#L345

Let me know if I misunderstood your question.

Kim-mins, 2023-12-11:
Thank you for the detailed response @evshary!

Actually, I'm suffering an issue on using Autoware.
I also reported this issue to autoware.universe repository, but I want to be sure that the bridge is not the cause.

Here's the video for the situation: [[video](https://drive.google.com/file/d/1wvIjMebNSS8q99lCK0UlYuInYzjA-HuS/view?usp=sharing)]
The two lines of the graph in the video shows a change(I plotted the accumulated data on every 20 times of messages) of `acceleration` and `speed` values from `/control/command/control_cmd` of Autoware messsage.

In the video, the ego vehicle brakes two times, at 0:31 and 1:17(last frame of the video).
For the frame at 0:31, the `acceleration` value goes down below -4, and the ego vehicle makes a really hard brake.
However, at 1:17, the `acceleration` value also goes down below -4, but the ego vehicle does not stops well enough and eventually collides with the NPC Vehicle.
The only difference of each frame is the road, whether the car is on the downhill(1:17) or not(0:31).

So, I have a question that, if the bridge has any dependency on the slope(downhill or not, ...) for handling control values(`acceleration`, `speed`, ...).
The reason could be that I'm using a bit old version(20230715) of your bridge, but I think the logic did not changed a lot, at least for handling control command.
I strongly guess the bridge has no difference on handing control values, cuz I think handling a slope is a functionality of Autoware side..

Thank you a lot! I'm using the bridge really well!

evshary, 2023-12-11:
To be honest, I'm afraid I can't provide more insights on the issue. I don't think the control mechanism in the bridge does anything with the slope (It just transforms messages between Carla and Autoware), but I'm not quite sure. It needs more investigation.
Does this only happen while driving on the slope? And does it always happen or not every time?
By the way, can you provide the link to the issue you open? I'm also interested in the issue, although I can't dive into it recently. (traveling now)

Kim-mins, 2024-02-14:
The issue above is resolved now.
It was the problem of the Autoware side.
Thank you for your help!

closed 2024-02-14 by Kim-mins

evshary, 2024-02-15:
Glad to know that it's fixed in the latest version.