evshary/autoware_carla_launch #48: Cannot run the most recent version of Autoware with the bridge

Kim-mins, 2023-12-12:
Hi @evshary!

I'm currently trying to use the most recent version of Autoware, but I encountered the following error while running `./script/run-autoware.sh`.

```1702373501.2178509 [INFO] [launch]: All log files can be found below /home/mins/.ros/log/2023-12-12-18-31-41-217338-4973e83026ad-1118
1702373501.2179277 [INFO] [launch]: Default logging verbosity is set to INFO
1702373508.5440278 [ERROR] [launch]: Caught exception in launch (see debug for traceback): Included launch description missing required argument 'duplicated_node_checker_param_path' (description: 'no description given'), given: [map_path, ...
```

It seems the new feature `duplicated_node_checker` is added to the Autoware, but the path `duplicated_node_checker_param_path` is not set. Could you please tell me where to modify to handle this issue?

Thank you. Please answer me when you are available.

evshary, 2023-12-12:
Which branch do you refer to, main or humble?
Can you check the `git status` and see whether there is anything modified?
From the error log, I guess there is some version conflict with Autoware, so I would like to check this first.

Kim-mins, 2023-12-13:
Thank you for the response @evshary!

Sorry for omitting the branch. I'm currently running on `humble` branch.
I also tested the branches `main` and `humble-0.9.14`, but they also have the same problem.

And I'm running Autoware with the latest(20231201) docker image from [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/153729759?tag=humble-20231201-prebuilt-cuda).

evshary, 2023-12-13:
OK, that makes sense. The bridge hasn't been tested with the latest Autoware.
If you want to try maybe you need to update the autoware_launch to the 20231201 first
https://github.com/evshary/autoware_launch/tree/carla_202308
And then update the commit here
https://github.com/evshary/autoware_carla_launch/blob/humble-0.9.14/autoware_carla.repos#L9
I haven't tried this, but some issues might need to be fixed.

Kim-mins, 2023-12-14:
Thank you for the reply @evshary!

I've done everything you suggested, and it seems there's no error regarding the parameters.
However, I got another weird error during the initialization of rviz:
```
[rviz2-87] [ERROR] [1702515332.095774043] [rviz2]: Failed to create an OpenGL context. BadValue (integer parameter out of range for operation)
[rviz2-87] [ERROR] [1702515332.095825124] [rviz2]: RenderingAPIException: Unable to create a suitable GLXContext in GLXContext::GLXContext at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXContext.cpp (line 60)
[rviz2-87] [ERROR] [1702515332.095855878] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Unable to create a suitable GLXContext in GLXContext::GLXContext at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXContext.cpp (line 60)
```

It seems there are some issue on NVIDIA driver, and `nvidia-smi` also does not work (emits `Failed to initialize NVML: Driver/library version mismatch`).
I guess the required version of driver is changed for the latest autoware, so I'm going to ask about it to Autoware team.

Thank you for your help! Please let me know if you know anything about this issue!

Kim-mins, 2023-12-21:
Hi @evshary!

I updated my cuda version and I tried to run the Autoware with docker image 20231215.
Following your answer, I also synchronized `autoware_launch`(I cloned the last commit in 20231215 of `autowarefoundation/autoware_launch` and applied your one commit to it) with the Autoware version and I'm on `humble` branch of the bridge (carla version is 0.9.13).

But I encountered another problem, and I could not connect bridge with carla.
Here's the `launch.log` and the image of rviz: [[launch.log](https://drive.google.com/file/d/1rbpQGV5zFAjUKExZBwYQcBVZ5cRiBDog/view?usp=sharing)] [[rviz image](https://drive.google.com/file/d/18UGrLgyXut5S3eNeKY5t8A1mMtNy_LUb/view?usp=sharing)]

From `launch.log`, I could check the error messages which have not been observed from the old version of Autoware(20230715):
```
1703148674.5888481 [ERROR] [lidar_centerpoint_node-47]: process has died [pid 4230, exit code -11, cmd '/autoware/install/lidar_centerpoint/lib/lidar_centerpoint/lidar_centerpoint_node --ros-args -r __node:=lidar_centerpoint -r __ns:=/perception/object_recognition/detection/centerpoint -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 --params-file /tmp/launch_params_4kszj1cz --params-file /tmp/launch_params_ywudzhz6 --params-file /tmp/launch_params_h9fkivzk --params-file /tmp/launch_params_k5u7sqlx --params-file /tmp/launch_params_99ao8xp4 --params-file /tmp/launch_params_swoe7dq0 --params-file /tmp/launch_params_h18_yoyd --params-file /tmp/launch_params_f1_mr26u --params-file /tmp/launch_params_1f5goik4 --params-file /tmp/launch_params_n6dblkl2 --params-file /home/mins/autoware_carla_launch/install/autoware_launch/share/autoware_launch/config/perception/object_recognition/detection/lidar_model/centerpoint_tiny.param.yaml --params-file /home/mins/autoware_carla_launch/install/autoware_launch/share/autoware_launch/config/perception/object_recognition/detection/lidar_model/detection_class_remapper.param.yaml -r ~/input/pointcloud:=/sensing/lidar/concatenated/pointcloud -r ~/output/objects:=objects'].
1703148675.2355673 [component_container_mt-58] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +888, GPU -4, now: CPU 1012, GPU 1290 (MiB)
1703148675.2513077 [component_container_mt-58] Could not open file /home/mins/autoware_data/traffic_light_classifier/traffic_light_classifier_mobilenetv2_batch_6.onnx
1703148675.2514405 [component_container_mt-58] Could not open file /home/mins/autoware_data/traffic_light_classifier/traffic_light_classifier_mobilenetv2_batch_6.onnx
1703148675.2514906 [component_container_mt-58] [E] [TRT] ModelImporter.cpp:733: Failed to parse ONNX model from file: /home/mins/autoware_data/traffic_light_classifier/traffic_light_classifier_mobilenetv2_batch_6.onnx
1703148675.2515256 [component_container_mt-58] [E] [TRT] Failed to parse onnx file
1703148675.2638552 [component_container_mt-58] [E] [TRT] [network.cpp::getInput::1968] Error Code 3: API Usage Error (Parameter check failed at: optimizer/api/network.cpp::getInput::1968, condition: index < getNbInputs()
1703148675.2640448 [component_container_mt-58] )
1703148675.5038345 [ERROR] [component_container_mt-58]: process has died [pid 4360, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=traffic_light_node_container -r __ns:=/perception/traffic_light_recognition/camera6 -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

Actually, I cannot sure whether the bridge or the Autoware makes the error, or maybe I made a mistake while updating `autoware_launch` version.
But I just want to know if you know any hint.

I always appreciate to your help! Thank you for reading!

evshary, 2023-12-25:
Hi @Kim-mins 
Maybe you can check whether the [planning_simulator](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) works well first.
If yes, then the error should come from the bridge.
BTW, I found that the machine learning models are now put inside the `~/autoware_data`, which did not exist before.
https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/artifacts
I guess this is the root cause.

Kim-mins, 2023-12-26:
Thank you for the information @evshary!!
I found the planning simulation does not works well too.
I also tried downloading `autoware_data`, but the planning simulation and the Autoware with the bridge did not work.
So I just asked to Autoware team. I'll let you know if I get any conclusion. Thank you!

Kim-mins, 2023-12-28:
Hi @evshary,

I can successfully run the planning simulation with the latest Autoware, but I could not run the latest Autoware with the bridge..
The setting is the same as before, and it seems the lanelet map is not loaded well.

Here's the screenshot:
[image]

As you can see, the pointcloud is loaded well but there's no lines regarding lanelet file. So the autoware fails to localize.

I think this situation is similar to that we suffered on July(when tried to run Autoware version 20230715), so I checked the parameter `lanelet2_map_projector_type` in `autoware_launch/config/map/lanelet2_map_loader.param.yaml`.
I tried `MGRS`, `local` and `UTM`, but every parameters did not work..

Do you have any idea on this issue?

Thank you!

evshary, 2024-01-12:
Hi @Kim-mins 
Sorry for the late reply. I'm working on another project recently.
I guess there might be changed a lot in Autoware from 0715 to 1215, and maybe there are some renamed topics or something else.
Could you share the log about this issue?
If I have time (I hope so...), I'll try to upgrade the Autoware as well.

Kim-mins, 2024-01-15:
Thank you for the response @evshary!!

I just found the (prebuilt) Autoware's latest version is 20240115 [[link](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe)].
I'll try the bridge on the latest version and share the log and the results soon.

Thank you!

Kim-mins, 2024-01-16:
Hi @evshary

I tried from the prebuilt Autoware docker image [[20240115](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/166938637?tag=humble-20240115-prebuilt-cuda-arm64)], and I failed to run the Autoware with the bridge.
For autoware launch, I tried the commit [[10b95d7669ff77a8fce3954c0b314e1126f92d9c](https://github.com/autowarefoundation/autoware_launch/commit/10b95d7669ff77a8fce3954c0b314e1126f92d9c)] and applied your [[last commit](https://github.com/autowarefoundation/autoware_launch/compare/main...evshary:autoware_launch:carla_20230715)].
As you mentioned before, I also downloaded the artifacts [[here](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/artifacts)].

Here's the launch.log file: [[launch.log](https://drive.google.com/file/d/1D61HQXM0KKsgrH1P-1c5XqWmKWCzmJ2e/view?usp=sharing)]
The situation of rviz is the same as the [comment above](https://github.com/evshary/autoware_carla_launch/issues/48#issuecomment-1871062128), and it seems the Autoware fails to find the initial location.

I cannot sure, but I could find the two suspicious parts:
1. Error log on `HW_ID`
```
1705367914.2826617 [component_container-74] [WARN] [1705367914.278349422] [control.control_validator]: diagnostic_updater: No HW_ID was set. This is probably a bug. Please report it. For devices that do not have a HW_ID, set this value to 'none'. This warning only occurs once all diagnostics are OK. It is okay to wait until the device is open before calling setHardwareID.
```

2. Error log on NDT align
```
1705367922.7957239 [pose_initializer_node-35] [INFO] [1705367922.794700570] [localization.util.pose_initializer_node]: Call NDT align server.
1705367922.7961497 [ndt_scan_matcher-34] [INFO] [1705367922.795110677] [localization.pose_estimator.ndt_scan_matcher]: waiting response
1705367922.7962029 [ndt_scan_matcher-34] [INFO] [1705367922.795243927] [localization.pose_estimator.ndt_scan_matcher]: Update map (Add: 0, Remove: 0)
1705367922.7965693 [ndt_scan_matcher-34] [INFO] [1705367922.795260143] [localization.pose_estimator.ndt_scan_matcher]: Skip map update
1705367922.7966206 [ndt_scan_matcher-34] [WARN] [1705367922.795267455] [localization.pose_estimator.ndt_scan_matcher]: No InputTarget. Please check the map file and the map_loader service
1705367922.7969539 [pose_initializer_node-35] [INFO] [1705367922.795325841] [localization.util.pose_initializer_node]: NDT align server failed.
1705367922.7973862 [service_log_checker-4] [ERROR] [1705367922.795475120] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/localization/util/pose_initializer_node)
1705367922.7974389 [service_log_checker-4] [ERROR] [1705367922.795874147] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
1705367922.7979085 [service_log_checker-4] [ERROR] [1705367922.795930542] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
1705367922.7979603 [service_log_checker-4] [ERROR] [1705367922.796016688] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/localization/util/default_ad_api/helpers/automatic_pose_initializer)
1705367923.6933978 [component_container-74] [INFO] [1705367923.692724870] [control.trajectory_follower.controller_node_exe]: Waiting for trajectory.
1705367923.6935482 [component_container-74] [INFO] [1705367923.692776002] [control.trajectory_follower.controller_node_exe]: Control is skipped since input data is not ready.
1705367924.2941697 [path_distance_calculator_node-82] [WARN] [1705367924.293058714] [autoware_api.utils.path_distance_calculator]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```
I don't know the details well, but I hope these could be the help.

I always appreciate to your help! Thank you!

evshary, 2024-01-19:
This issue should come from the latter one, which means we need to take a look at why `NDT align server failed`.
I guess there is something wrong with the sensor transformation.
Maybe checking the output topic of sensors (GNSS & pointcloud) is a good starting point.
According the node diagram, the input of ndt_scan_matcher might have some problem.
[image]

Kim-mins, 2024-01-23:
Thank you for the detailed response @evshary!

For debugging, I focused on the log message below:
```
1705367922.7962029 [ndt_scan_matcher-34] [INFO] [1705367922.795243927] [localization.pose_estimator.ndt_scan_matcher]: Update map (Add: 0, Remove: 0)
```

For the comparison, I checked if I can get the same log message from 20230715, but I could get a bit different log as below:
```
1705982206.5634191 [ndt_scan_matcher-31] [INFO] [1705982206.561748232] [localization.pose_estimator.ndt_scan_matcher]: Update map (Add: 1, Remove: 0)
```
**which states the added one is `1`, instead of `0`.**

So I checked the source code(I'm not that familiar though..), and I could find [the code line relevant to the log above](https://github.com/autowarefoundation/autoware.universe/blob/1aee603e258e391825340c0f847ddd314943c758/localization/ndt_scan_matcher/src/map_update_module.cpp#L80), and the variable related to that `1` or `0` is **`maps_to_add`**, which is the parameter of the function `update_ndt`.

The function `update_ndt` is called at [line 76 of the same code](https://github.com/autowarefoundation/autoware.universe/blob/1aee603e258e391825340c0f847ddd314943c758/localization/ndt_scan_matcher/src/map_update_module.cpp#L76), and I thought the parameter related to the variable `maps_to_add` have to do with pointcloud.

So I checked the message(as you recommended) from `/localization/util/downsample/pointcloud`, which is related to the pointcloud, **and I could not get any message from the topic.**
Also, following the topology of the diagram you provided, I echoed every topics (topologically) above the `ndt_scan_matcher`, but I cannot get any message from the **latest version** (I echoed to the topics `/localization/util/voxel_grid_downsample/pointcloud`, `/localization/util/measurement_range/pointcloud`, `/sensing/lidar/top/pointcloud`**).

However, when I echo the topic `/localization/util/downsample/pointcloud` on 20230715 version, I could get the message as below:
```
header:
  stamp:
    sec: 86
    nanosec: 608223673
  frame_id: base_link
height: 1
width: 1072
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
is_bigendian: false
point_step: 16
row_step: 17152
data:
- 61
- 6
- 60
- 64
- 204
- 167
...
- '...'
is_dense: true
```

I also echoed to the (topologically) above topics on 20230715, but **I could get the message from all the topics above**, except `/sensing/lidar/top/pointcloud`, which seems not a published topic of the old version.

By the way, I could find some topics in the diagram you provided **are not in my topic list**(from `ros2 topic list`).
Here are the list of the topics I cannot find:
```
/localization/pose_twist_fusion_filter/pose_with_covariance_no_yaw_bias (from ekf_localizer to ndt_scan_matcher (in the diagram))
/sensing/lidar/*/velodyne_packets
/sensing/lidar/*/self_cropped/pointcloud
/sensing/lidar/*/mirror_cropped/pointcloud
...(maybe more..)
```
I wonder if those topics are processed internally, so that I cannot listen to it from outside.
I inform you just with the concern that the topics above are new and should be handled.

Thank you for reading, even though you are so busy now!
I always appreciate to your hard work!

evshary, 2024-01-29:
@Kim-mins Thank you for diving into the code!
I agree with you it's weird that `localization/util/downsample/pointcloud` disappeared in the latest Autoware version.
It is worth a deeper investigation.
About the the lacking of
```
/sensing/lidar/*/velodyne_packets
/sensing/lidar/*/self_cropped/pointcloud
/sensing/lidar/*/mirror_cropped/pointcloud
```
We don't need this in Carla, since we can just transformed the lidar topic from Carla into `/sensing/lidar/concatenated/pointcloud` directly.
https://github.com/evshary/autoware_carla_launch/blob/main/src/carla_pointcloud/src/carla_pointcloud/carla_pointcloud_interface_node.cpp#L37C83-L37C106

However, I did find something changed in the latest Autoware
https://github.com/autowarefoundation/autoware.universe/commit/3e62ad4e6d5299c089d11a44a9373d5854b48c81#diff-03009c8cf043839bf6bc8604015a841a31b3e3981d91fb0e6f1f58c8558446c9L17
Would you mind modifying the code here with `/sensing/lidar/top/pointcloud` and test again?
https://github.com/evshary/autoware_carla_launch/blob/main/src/carla_pointcloud/src/carla_pointcloud/carla_pointcloud_interface_node.cpp#L35

Kim-mins, 2024-01-29:
Thank you for the response @evshary!!

I tried your suggestion by modifying the code
```cpp
velodyne_points_localization =
            this->create_publisher<sensor_msgs::msg::PointCloud2>("/sensing/lidar/top/outlier_filtered/pointcloud", rclcpp::SensorDataQoS());
```
to
```cpp
velodyne_points_localization =
            this->create_publisher<sensor_msgs::msg::PointCloud2>("/sensing/lidar/top/pointcloud", rclcpp::SensorDataQoS());
```
whose string `/outlier_filtered` is removed, and built the bridge and autoware again.

However, I got the same situation, with the same log(`NDT align server failed`) as below:
```
1706514376.2651873 [pose_initializer_node-35] [INFO] [1706514376.263545527] [localization.util.pose_initializer_node]: EKF Deactivation succeeded
1706514376.2676754 [pose_initializer_node-35] [INFO] [1706514376.266079588] [localization.util.pose_initializer_node]: NDT Deactivation succeeded
1706514376.2694700 [pose_initializer_node-35] Failed to find match for field 'x'.
1706514376.2695730 [pose_initializer_node-35] Failed to find match for field 'y'.
1706514376.2696171 [pose_initializer_node-35] Failed to find match for field 'z'.
1706514376.2696538 [pose_initializer_node-35] [INFO] [1706514376.267910920] [localization.util.pose_initializer_node]: Call NDT align server.
1706514376.2701447 [ndt_scan_matcher-34] [INFO] [1706514376.268579316] [localization.pose_estimator.ndt_scan_matcher]: waiting response
1706514376.2702029 [ndt_scan_matcher-34] [INFO] [1706514376.268879353] [localization.pose_estimator.ndt_scan_matcher]: Update map (Add: 0, Remove: 0)
1706514376.2702434 [ndt_scan_matcher-34] [INFO] [1706514376.268904988] [localization.pose_estimator.ndt_scan_matcher]: Skip map update
1706514376.2702796 [ndt_scan_matcher-34] [WARN] [1706514376.268913923] [localization.pose_estimator.ndt_scan_matcher]: No InputTarget. Please check the map file and the map_loader service
1706514376.2708511 [pose_initializer_node-35] [INFO] [1706514376.268995691] [localization.util.pose_initializer_node]: NDT align server failed.
1706514376.2713027 [service_log_checker-4] [ERROR] [1706514376.269196018] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/localization/util/pose_initializer_node)
1706514376.2716570 [service_log_checker-4] [ERROR] [1706514376.269990052] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
1706514376.2717092 [service_log_checker-4] [ERROR] [1706514376.270057784] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
1706514376.2717612 [service_log_checker-4] [ERROR] [1706514376.270159223] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/localization/util/default_ad_api/helpers/automatic_pose_initializer)
1706514376.5615833 [path_distance_calculator_node-82] [WARN] [1706514376.560890805] [autoware_api.utils.path_distance_calculator]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```

If you have time and more suggestions for the fix, I think you can provide me more, since I have the running environment now.

I'll also investigate more about the modified topics. Thank you!

Kim-mins, 2024-01-29:
Hi @evshary

I cannot sure, but the commit below seems introducing a new topic regarding sensing `input_twist_topic_type`.
- https://github.com/autowarefoundation/autoware.universe/commit/68ee0f220fcd3e4f0ed9276270008d04fd579120

evshary, 2024-01-29:
> * [autowarefoundation/autoware.universe@68ee0f2](https://github.com/autowarefoundation/autoware.universe/commit/68ee0f220fcd3e4f0ed9276270008d04fd579120)
> * [autowarefoundation/autoware.universe@b498c53](https://github.com/autowarefoundation/autoware.universe/commit/b498c5352390e3637fb723140bf7b541d5bb04a8)
> * [autowarefoundation/autoware.universe@8a13959](https://github.com/autowarefoundation/autoware.universe/commit/8a139596de1ca20318cbb8d08a1a569e86988a90)

@Kim-mins Thank you for the investigation.
After a quick view, I don't think these modifications make any difference, but maybe I missed something.
Will take a further look later.

> I also checked the topics above that did not work before(/localization/util/voxel_grid_downsample/pointcloud, /localization/util/measurement_range/pointcloud, /sensing/lidar/top/pointcloud, /sensing/lidar/top/pointcloud), and I could observe the output from every topics.

Sounds good. At least we have some progress. Since lidar data should be good now, could you help me check whether GNSS works well or not?

Kim-mins, 2024-01-30:
Thank you for the response @evshary!

For GNSS, I echoed the topic with the string `gnss`(please tell me if there's more), and here's the echo result for each topic relevant to GNSS (**every topic works**)
- `/sensing/gnss/fixed`:
    ```
    stamp:
      sec: 153
      nanosec: 208064745
    data: true
    ```
- `/sensing/gnss/pose`:
    ```
    header:
      stamp:
        sec: 136
        nanosec: 258064492
      frame_id: map
    pose:
      position:
        x: 66022.95483240287
        y: 99706.26651246473
        z: 19.40150260925293
      orientation:
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    ```
- `/sensing/gnss/pose_with_covariance`
    ```
    header:
      stamp:
        sec: 174
        nanosec: 808065067
      frame_id: map
    pose:
      pose:
        position:
          x: 66022.95483240287
          y: 99706.26651246473
          z: 19.40150260925293
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
      covariance:
      - 10.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 10.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 10.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
      - 0.0
    ```
- `/sensing/gnss/ublox/nav_sat_fix`
    ```
    header:
      stamp:
        sec: 186
        nanosec: 708065244
      frame_id: gnss_link
    status:
      status: 1
      service: 15
    latitude: -0.002653832465426831
    longitude: 1.3563788603466434e-05
    altitude: 19.40150260925293
    position_covariance:
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    position_covariance_type: 0
    ```

It is good signal since every topic work, but I saw a bit weird `position` from `/sensing/gnss/pose` and `/sensing/gnss/pose_with_covariance`:
```
position:
    x: 66022.95483240287
    y: 99706.26651246473
    z: 19.40150260925293
```

When I echoed to the topic `/sensing/gnss/pose` and `/sensing/gnss/pose_with_covariance` from the version `20230715`, I could get the position values as below:
```
position:
    x: 40.15022040623561
    y: -143.9113872943367
    z: -0.20792973165465334
```
which are valid for the given map(from Carla).

I thought this could be the map configuration error that we suffered while running `20230715`, so I tried modifying map projection types(`local`, `MGRS`, `LocalCartesianUTM`, **the result above is from `local`**), following the instructions [here](https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_projection_loader/README.md).

When I tried to echo to the topics relevant to GNSS for each projection type, I could observe that, two of GNSS topics(`/sensing/gnss/fixed`, `/sensing/gnss/pose`) are invisible with the projection type `MGRS` and `LocalCartesianUTM`, and the rest of the visible topics do not publish any message.

So I think the projection type `local` is the correct one, and maybe I should check another configurations.. do you have any idea on this?

Thank you for reading and thank you for your help!

Kim-mins, 2024-01-30:
Hi @evshary!

From the [map loader instruction](https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_projection_loader/README.md), I could see the limitation of using `local` as the projection type, which was updated last week.
[image]

I think the **limitation** is exactly what we are suffering now(GNSS Localization), so maybe the map projection type is the root cause of our issue. Do you have any idea on this?

Kim-mins, 2024-01-30:
Or.. I also checked the `launch.log` again and I focused on the warning on `ekf_localizer`.
```
[ekf_localizer-38] [WARN] [1706593199.330410044] [localization.pose_twist_fusion_filter.ekf_localizer]: The node is not activated. Provide initial pose to pose_initializer
```
According to the diagram you provided before, the component `pose_initializer` gets input from two topics, `/sensing/gnss/pose_with_covariance` and `/initialpose`.
However, I could not get the message from `/initialpose`, which is from `HMI`, so maybe this could be the another root cause. Do you have any idea on this?

evshary, 2024-02-02:
`/initialpose` should not be the issue, since we don't need users to provide the initial pose. GNSS should provide enough information.
The issue of the map loader might be the key, although I haven't taken a deeper look.
Maybe we can focus on this first.

evshary, 2024-02-07:
@Kim-mins Maybe the problem is related to the issue
https://github.com/orgs/autowarefoundation/discussions/3759#discussioncomment-6862856

Kim-mins, 2024-02-07:
Thank you for the response @evshary!

I checked the comment you mentioned and it seems they decided not to support some of the `projector_type`s and change the way to provide the `projector_type`.

I'm using `projector_type` as `local` since this is the only option that loads lanelet2 map correctly. (I referred to [the code](https://github.com/tkhmy/autoware.universe/blob/0443a08de439aab33457cf599170f6da076d487c/map/map_loader/src/lanelet2_map_loader/lanelet2_map_loader_node.cpp#L104) and it seems the option `local` is valid.)
I thought I'm providing projection type properly, following the [map projection loader documentation](https://github.com/autowarefoundation/autoware.universe/tree/main/map/map_projection_loader), but maybe I should check it more.

By the way, after some investigation, I observed that **the `gnss_poser` node is not loaded** with the `projection type` `local` but the node is loaded well with the `projection type` `MGRS`. `gnss_poser` is loaded well with the version 20230715, so maybe this could be the root cause of our issue. I'm focusing on this point now. Could you please share your opinion on this?

evshary, 2024-02-07:
Is there any more log from `gnss_poser` before it died?
Maybe we need to add some debug messages into the `gnss_poser` to see what happened.
https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/gnss_poser/src/gnss_poser_core.cpp#L29
Are you using the prebuilt Autoware? Or do you build the Autoware by yourself?

Kim-mins, 2024-02-07:
Thank you for the response @evshary!

Unfortunately, there's no more log from `gnss_poser`. The only logs I could get are the two lines below (the node dies just after being turned on):
```
...
1707289879.2995105 [INFO] [gnss_poser-36]: process started with pid [610]
1707289879.2996364 [ERROR] [gnss_poser-36]: process has died [pid 610, exit code -6, cmd '/autoware/install/gnss_poser/lib/gnss_poser/gnss_poser --ros-args -r __node:=gnss_poser -r __ns:=/sensing/gnss -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 --params-file /autoware/install/gnss_poser/share/gnss_poser/config/gnss_poser.param.yaml -r fix:=ublox/nav_sat_fix -r autoware_orientation:=/autoware_orientation -r gnss_pose:=pose -r gnss_pose_cov:=pose_with_covariance -r gnss_fixed:=fixed'].
...
```

I'm trying on the prebuilt Autoware from docker now(version 20240201).
I guess you want to check if I can append the code for debugging and build the source code.
I think I can anyway, but I have no idea on adding debugging options for `gnss_poser`. Could you please help me?

evshary, 2024-02-08:
Hi @Kim-mins 

Thank you for the instant check with the Autoware team.
I'm off from the office these days and might not be able to take a look.
Starting from checking how to use another projector type should be a more feasible way to solve the problem.

Kim-mins, 2024-02-09:
Thank you for the response @evshary!

I'm discussing the way to use another projector type with the Autoware team now.
I'll let you know if I got to know how.
Have a nice holiday!

Kim-mins, 2024-02-15:
Hi @evshary!

After some discussion with the Autoware team, I changed the projector type to `LocalCartesianUTM ` or `TransverseMercator` and the `gnss_poser` works.

However, I could get the message from the topic `/sensing/gnss/pose` as below:
```
header:
  stamp:
    sec: 643
    nanosec: 698663330
  frame_id: map
pose:
  position:
    x: 88.70719127816847
    y: -273.9020325280726
    z: 21.801562213897704
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
```
which is slightly different with the real position of the ego vehicle in the simulator, whose position is `x=88.619972, y=275.477936, z=0.001562`.

I think I should change the `latitude`, `longitute` and `altitude` in the `map_projector_info.yaml` to correct the offset, but I cannot know how. Do you have any idea for this issue?
For more information, you can refer to the discussion: https://github.com/orgs/autowarefoundation/discussions/4157#discussioncomment-8449828

Thank you! Enjoy your holiday and please answer me when you are available.

evshary, 2024-02-15:
@Kim-mins Thank you for the information. I know little about the issue.
I'll set up the environment and see how to reproduce the issue.

Kim-mins, 2024-02-15:
Thank you @evshary!

Here's the information for the environment setting (the steps I took):

1. Pull the latest docker and update `autoware_launch` to the corresponding version. Also the image name of the Dockerfile `./container/Dockerfile_autoware` should be modified. I'm using the version `20240201` now.

2. Download `autoware_data` from https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/artifacts (you told me:) ) and place them to the `$HOME/autoware_data` directory.

3. Modify the last line of `./container/run-autoware-docker.sh` to the line below: (add mounting `autoware_data`)
```
rocker --nvidia --privileged --x11 --user --volume $(pwd):$HOME/autoware_carla_launch --volume $HOME/autoware_data:$HOME/autoware_data -- ${DOCKER_IMAGE}
```

4. Modify the topic for the localization from `/sensing/lidar/top/outlier_filtered/pointcloud` to `/sensing/lidar/top/pointcloud` (you told me too): https://github.com/evshary/autoware_carla_launch/blob/main/src/carla_pointcloud/src/carla_pointcloud/carla_pointcloud_interface_node.cpp#L35

5. For each map in `carla_map/*`, I added `map_projector_info.yaml`, following the instruction here: https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_projection_loader/README.md
```
# File `autoware_carla_launch/carla_map/Town01/map_projector_info.yaml`
projector_type: TransverseMercator
vertical_datum: WGS84
map_origin:
  latitude: 0.0 # [deg]
  longitude: 0.0 # [deg]
  altitude: 0.0 # [m]
```
For the `projector_type`, only `LocalCartesianUTM` and `TransverseMercator` work now.

I hope these would help you!

Kim-mins, 2024-02-15:
Hi @evshary!

I'm still on discussion with the team now, and I got a possible cause from the team: https://github.com/orgs/autowarefoundation/discussions/4157#discussioncomment-8476110

He saids, maybe the position(the center of the ego vehicle) in the simulator is different from the `base_link` of Autoware.

I think this was not the issue when we used `local` as a `projector_type`, but we changed it now. So I think this could be the cause of our issue.
This is just my opinion, and this would have nothing to do with our issue. Sorry if it's not the problem.

Thank you!

Kim-mins, 2024-02-21:
Hi @evshary!

I think the error is from the concatenation error on pointcloud.
For the **old version**, I could check that the pointcloud concatenation is successful by the message below(from the topic `/diagnostics`):
```
header:
  stamp:
    sec: 57
    nanosec: 362018021
  frame_id: ''
status:
- level: "\0"
  name: 'concat_downsampled_pcl: concat_status'
  message: Concatenate all topics
  hardware_id: concatenate_data_checker
  values:
  - key: long_range/pointcloud
    value: OK
  - key: outlier_filter/pointcloud
    value: OK
```

However, the **recent version** makes an error for the concatenation:
```
header:
  stamp:
    sec: 55
    nanosec: 92242581
  frame_id: ''
status:
- level: "\x01"
  name: 'concatenate_data: concat_status'
  message: Some topics are not concatenated
  hardware_id: concatenate_data_checker
  values:
  - key: /sensing/lidar/top/pointcloud
    value: OK
  - key: /sensing/lidar/left/pointcloud
    value: NG
  - key: /sensing/lidar/right/pointcloud
    value: NG
```

And I also got the log below:
[image]

When I checked the data from `/sensing/lidar/right/pointcloud` and `/sensing/lidar/left/pointcloud`, they do not publish anything, and it seems that we need those data for the concatenation.

Do you have any idea for this?

Thank you!

Kim-mins, 2024-02-21:
Sorry for bothering, but could you please refer to the last 3 comments here, which could be the hint for the error? Maybe you could get some more information https://github.com/orgs/autowarefoundation/discussions/4157#discussioncomment-8539566

Thanks!

Kim-mins, 2024-02-21:
Hi @evshary!

After a long discussion with the team, I got to know that [this PR](https://github.com/autowarefoundation/autoware.universe/pull/6091) made the pose initialization fail.

The PR changes the container name and it eventually made component name unsynced with ours, which is 
- Latest Autoware: /sensing/lidar/*/pointcloud_preprocessor/**pointcloud_container** 
- Ours: /sensing/lidar/*/pointcloud_preprocessor/**velodyne_node_container**

Since the PR was merged on 2024.01.19, I ran Autoware 20240115 with the prebuilt docker and the localization is successful. So maybe that PR is the key.

I also tried the latest version by changing [here](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/f8c08777721e04c7f06826755ea30134f22a4a2b/common_sensor_launch/launch/velodyne_VLS128.launch.xml#L16C14-L16C28) and [here](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/f8c08777721e04c7f06826755ea30134f22a4a2b/common_sensor_launch/launch/velodyne_VLP16.launch.xml#L16), but it failed. Here's the printed component with the command `ros2 component list`:
```
/autoware_api/external/rtc_controller/container
  1  /autoware_api/external/rtc_controller/node
/planning/scenario_planning/motion_velocity_smoother_container
  1  /planning/scenario_planning/motion_velocity_smoother
  2  /planning/scenario_planning/glog_component
/control/control_container
  1  /control/external_cmd_selector
  2  /control/trajectory_follower/controller_node_exe
  3  /control/external_cmd_converter
  4  /control/control_validator
  5  /control/trajectory_follower/lane_departure_checker_node
  6  /control/shift_decider
  7  /control/vehicle_cmd_gate
  8  /control/operation_mode_transition_manager
  9  /control/glog_component
/system/mrm_emergency_stop_operator/mrm_emergency_stop_operator_container
  1  /system/mrm_emergency_stop_operator
/sensing/lidar/right/pointcloud_preprocessor/pointcloud_container
/planning/scenario_planning/lane_driving/motion_planning/motion_planning_container
  1  /planning/scenario_planning/lane_driving/motion_planning/glog_component
  2  /planning/scenario_planning/lane_driving/motion_planning/elastic_band_smoother
  3  /planning/scenario_planning/lane_driving/motion_planning/obstacle_avoidance_planner
  4  /planning/scenario_planning/lane_driving/motion_planning/obstacle_velocity_limiter
  5  /planning/scenario_planning/lane_driving/motion_planning/obstacle_stop_planner
/planning/mission_planning/mission_planner_container
  1  /planning/mission_planning/mission_planner
  2  /planning/mission_planning/glog_component
/perception/traffic_light_recognition/traffic_light_arbiter/container
  1  /perception/traffic_light_recognition/traffic_light_arbiter/arbiter
/sensing/lidar/left/pointcloud_preprocessor/pointcloud_container
/system/mrm_comfortable_stop_operator/mrm_comfortable_stop_operator_container
  1  /system/mrm_comfortable_stop_operator
/default_ad_api/container
  1  /default_ad_api/node/autoware_state
  2  /default_ad_api/node/fail_safe
  3  /default_ad_api/node/interface
  4  /default_ad_api/node/localization
  5  /default_ad_api/node/motion
  6  /default_ad_api/node/operation_mode
  7  /default_ad_api/node/perception
  8  /default_ad_api/node/planning
  9  /default_ad_api/node/routing
  10  /default_ad_api/node/vehicle
  11  /default_ad_api/node/vehicle_info
/pointcloud_container
  1  /perception/object_recognition/detection/clustering/low_height_crop_box_filter
  2  /pointcloud_container/glog_component
  3  /perception/occupancy_grid_map/occupancy_grid_map_node
  4  /sensing/lidar/concatenate_data
  5  /perception/object_recognition/detection/centerpoint/lidar_centerpoint
  6  /perception/obstacle_segmentation/crop_box_filter
  7  /perception/object_recognition/detection/voxel_grid_downsample_filter
  8  /perception/object_recognition/detection/clustering/euclidean_cluster
  9  /perception/obstacle_segmentation/common_ground_filter
  10  /perception/object_recognition/detection/voxel_based_compare_map_filter
  11  /perception/obstacle_segmentation/occupancy_grid_map_outlier_filter
/perception/traffic_light_recognition/camera6/traffic_light_node_container
  1  /perception/traffic_light_recognition/camera6/traffic_light_image_decompressor
[... 62 lines cut ...]
  5  /awapi/force_available_relay
  6  /awapi/running_modules_relay
  7  /awapi/autoware_engage_relay
  8  /awapi/vehicle_engage_relay
  9  /awapi/put_route_relay
  10  /awapi/put_goal_relay
  11  /awapi/lane_change_approval_relay
  12  /awapi/force_lane_change_relay
  13  /awapi/external_approval_relay
  14  /awapi/force_approval_relay
  15  /awapi/obstacle_avoid_approval_relay
  16  /awapi/traffic_signal_relay
  17  /awapi/overwrite_traffic_signals_relay
  18  /awapi/speed_exceeded_relay
  19  /awapi/crosswalk_status_relay
  20  /awapi/intersection_status_relay
  21  /awapi/expand_stop_range_relay
  22  /awapi/pose_initialization_request_relay
```
which does not show any sub-component(?) for the nodes `/sensing/lidar/*/pointcloud_preprocessor/pointcloud_container`.

Do you know how to modify it?
I think the files [here](https://github.com/evshary/autoware_carla_launch/tree/main/src/carla_sensor_kit_launch) would be changed, but I don't know how.. Do you have any idea on this?

Thank you!

Kim-mins, 2024-02-22:
I investigated more, and I think the changes for `sample_sensor_kit` in [this commit](https://github.com/autowarefoundation/sample_sensor_kit_launch/commit/a6a7063e5c02e85513c7ce0288d72dde6475e286) and [this commit](https://github.com/autowarefoundation/sample_sensor_kit_launch/commit/5ccd754fde6a39d8ddb3a8ee2f0f0e8be540f6a5) seem to be applied to the [carla_sensor_kit](https://github.com/evshary/autoware_carla_launch/tree/main/src/carla_sensor_kit_launch). Could you please check it when you are available?

Thank you!

Kim-mins, 2024-03-04:
Hi @evshary,

I'm currently running Autoware 20240115, but it makes somewhat trivial errors so that I cannot play with it well.. could you please help we if you have a time?

Thank you.

evshary, 2024-03-04:
Hi @Kim-mins
I'll check that later. Sorry for the late reply. I'm working on another urgent project recently.

Kim-mins, 2024-03-04:
Sorry for mentioning when you are busy.
Your work is the first, and please check this issue only if you have a time.

Thank you!

evshary, 2024-09-27:
Now we can run with 2024.09 version on the main branch. Close the issue.

closed 2024-09-27 by evshary

Kim-mins, 2024-09-28:
@evshary Good news! Thank you for sharing!