evshary/autoware_carla_launch #49: Carla Map seems to be mismatched with the point cloud map in the Autoware ROS Humble version

littlevu, 2023-12-14:
Hi @evshary,

I am following this tutorial: https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/ to debug my autoware carla bridge code. This below command run very well in AWSim Map. However, when I replace AWSim Map to Carla Map Town 01, my rviz seems to be mismatched between lanelet2 map and point cloud map. Do you check it?

```bash
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=<your mapfile location>
```

Thank you.

evshary, 2023-12-25:
Hi @littlevu
I'm a little confused about your environment. Do you mean you tried to use Carla Map Town 01 in AWSIM?
If so, I think this will not work since you need to create your own Unity project which corresponds to the Carla Map.

evshary, 2024-03-17:
I'll close the issue and feel free to reopen it if you have any further questions.

closed 2024-03-17 by evshary