evshary/autoware_carla_launch #51: The "--position" argument in zenoh_carla_bridge does not work.

hwdryan, 2024-01-10:
Hi @evshary,

I modified the "script/run-bridge.sh" to add the "--pygame" and"--position" to the poetry command. The "--pygame" works fine. But when I use "--position", the script will get stuck, "main.py" will not respond and crash.

Here is the poetry command:
`"poetry -C ${PYTHON_AGENT_PATH} run python3 ${PYTHON_AGENT_PATH}/main.py \
                    --host ${CARLA_SIMULATOR_IP} --pygame --rolename ${VEHICLE_NAME} --position 396,105.4,0,0,0,0 \
                    2>&1 | tee ${LOG_PATH}/vehicle.log"`

Information from running the bash script:

> sleep 5 && RUST_LOG=z=info /home/weidonghu/autoware_carla_launch/external/zenoh_carla_bridge/target/release/zenoh_carla_bridge                     --zenoh-listen tcp/0.0.0.0:7447 2>&1 | tee bridge_log/2024-01-10_12:25:32//bridge.log
poetry -C /home/weidonghu/autoware_carla_launch/external/zenoh_carla_bridge/carla_agent run python3 /home/weidonghu/autoware_carla_launch/external/zenoh_carla_bridge/carla_agent/main.py                     --host 127.0.0.1 --pygame --rolename v1 --position 396,105.4,0,0,0,0                     2>&1 | tee bridge_log/2024-01-10_12:25:32//vehicle.log
INFO: listening to server 127.0.0.1:2000
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  130 (MIT-SHM)
  Minor opcode of failed request:  3 (X_ShmPutImage)
  Value in failed request:  0x500
  Serial number of failed request:  238
  Current serial number in output stream:  239
 INFO  zenoh_carla_bridge > Running Carla Autoware Zenoh bridge...
 INFO  zenoh::net::runtime > Using PID: 5fd6e2fa69c8e32a8dcdc836ccea73ac
 INFO  zenoh::net::runtime::orchestrator > Zenoh can be reached at: tcp/192.168.188.83:7447
 INFO  zenoh::net::runtime::orchestrator > Zenoh can be reached at: tcp/172.17.0.1:7447
 INFO  zenoh::net::runtime::orchestrator > zenohd listening scout messages on 224.0.0.224:7446
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.4.0-1-gb3fbf09-dirty 
WARNING: Simulator API version  = 0.9.13 
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.4.0-1-gb3fbf09-dirty 
WARNING: Simulator API version  = 0.9.13

From this information, I noticed that the sensors here are not created?

Similar thing happens when I use `client.get_world()` instead of `client.load_world()` in external/zenoh_carla_bridge/carla_agent/simulation/loop.py, and designate a vehicle I spawned in carla as the player. Although in this case, the pygame window works fine, the sensors doesn't work, causing the localization and carmera image in autoware RVIZ window not working. 

Can you help me with this? Thank you!

hwdryan, 2024-01-10:
The problem for the "--position" argument is resolved. It seems that the z value should be set a bit higher than 0.

But why using existing vehicle as player for autoware doesn't work is still in doubt. Thank you!

evshary, 2024-01-12:
Hi @hwdryan , is there any reason you want to use `get_world` instead of `load_world`?
Does that mean you've already had a Carla scenario and tried to bridge the existing vehicle to Autoware?
Also, can you share the log file for further analysis?

hwdryan, 2024-01-12:
Hi @evshary , thank you for replying. Yes, indeed I have a Carla scenario and try to use the ego vehicle in the scenario as AV in Autoware. This is part of my master thesis project. I have found a walk around, which is to launch the scenario inside the world loaded by Carla Automated bridge. 

I will upload the log files as soon as possible. Thank you!

evshary, 2024-03-17:
I'll close the issue and feel free to reopen it if you have any further questions.

closed 2024-03-17 by evshary

Justin-Xiang, 2024-11-26:
Hi @hwdryan, are you still working on it? I also want to use the ego vehicle in the scenario as AV in Autoware.