evshary/autoware_carla_launch #52: carla-sys Rust crate unable to build

MarioAlmelaUMH, 2024-01-25:
Hi, Evshary. During the last couple of days we experienced a specific issue with the building of the zenoh_carla_bridge. Apparently, the carla-sys Rust crate cannot build. The process gets frozen and the memory usage increases until it overflows the available memory and Make shows a 137 error. This happens on two different computers with 2 different bridge installations.

[image]

We tested it on a third computer that had an older version of the bridge and this time it worked fine. We discovered that the Rust crate **cxx** is most likely the cause. The computers that don't work both have version 1.0.115 of cxx, while the one that does work has version 1.0.104. The thing is we don't know how to change that crate's version. It doesn't appear as a dependency on the `cargo.toml` files and the only way to see it is by running `cargo tree` on the `zenoh_carla_bridge` folder. We think the cxx crate is installed along with the first installation of Rust, when `make prepare_bridge` is run in the container for the first time. As we cannot directly change the version, re-running the command does not fix the issue.

Do you know how we can revert the cxx crate's version back to 1.0.104?

Thank you for your time.
Best regards,
Mario

evshary, 2024-01-29:
@MarioAlmelaUMH Can you show the what you've modified in `zenoh_carla_bridge`?
I guess the `Cargo.lock` is changed in your case.
https://github.com/evshary/zenoh_carla_bridge/blob/main/Cargo.lock#L853
You can restore the `Cargo.lock` back to the default one and try again.

MarioAlmelaUMH, 2024-01-29:
Hi, @evshary . We solved it. After clearing the `Cargo.lock`and rebuilding many times, we discovered (very randomly) that the issue was in the `env.sh` file. The last line, `export RUST_LOG=info` was set to `debug` instead. Changing it back to `info` solved the problem. We don't know exactly why this connection exists between the Rust log level and the building process, but in our case it worked.

Best regards,
Mario

closed 2024-01-29 by MarioAlmelaUMH

evshary, 2024-01-29:
> Hi, @evshary . We solved it. After clearing the `Cargo.lock`and rebuilding many times, we discovered (very randomly) that the issue was in the `env.sh` file. The last line, `export RUST_LOG=info` [...]

Sounds a little weird, because RUST_LOG should only take effect on runtime.
Anyway, let me know if the issue still happens.

LLusvarghi, 2024-01-29:
Hi @evshary . I have been working with Mario on the bridge over the last couple of months. Nice to meet you. 
We agree with you that the RUST_LOG level should only take effect on runtime. However, for some unknown reason, the carla-sys crate cannot build if RUST_LOG=debug. Actually, we have been able to solve this problem after realizing that the same issue had already been raised on the [official carla-sys GitHub repo](https://github.com/jerry73204/carla-rust/issues/3).
Anyways, thank you for your prompt reply.

evshary, 2024-01-29:
@LLusvarghi Nice to meet you!
Thanks for your information. I missed that issue.
It's really interesting why the RUST_LOG is related to building...