evshary/autoware_carla_launch #53: Can Carla Simulation publish to local ROS 2 Autoware topics

dtonda8, 2024-01-25:
When I set a 2D Goal Pose on Rviz, I hope to see publishes to Autoware topics.
I am following the [Carla-Autoware Demo](https://autoware.org/running-multiple-autoware-powered-vehicles-in-carla-using-zenoh/) but cannot find any Autoware topics (i.e. /control/command/control_cmd) from `ros2 topic list`. I am using a single host for both the bridge and Autoware dockers.

I've tried running `ros2 topic list` in the "run-autoware-docker" container but could only find `/rosout` and `/parameter_events`

Am I missing something? 

Any help/direction will be appreciated.

dtonda8, 2024-01-25:
@evshary! I'd appreciate any advice:)

dtonda8, 2024-01-28:
I found an alternative that better suited my needs. I am now using [op_carla_bridge](https://github.com/orgs/autowarefoundation/discussions/2828#discussioncomment-7929757) and [op_bridge](https://github.com/hatem-darweesh/op_bridge/blob/ros2-humble/README.md). I won't close this issue since other's may want to know how to publish to local ROS 2 Autoware topics.

evshary, 2024-01-29:
Hi @dtonda8 
I'm glad you found another project that fits your needs.
About your question, maybe it's similar to [this issue](https://github.com/evshary/autoware_carla_launch/issues/25)
FYI

dtonda8, 2024-01-29:
Thanks @evshary. The issue you linked is the exact same issue as the one I posed (oops)

closed 2024-01-29 by dtonda8