evshary/autoware_carla_launch #60: Forbidden to pull image ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64

ChinYikMing, 2024-03-11:
How to reproduce?
```
sudo ./container/run-bridge-docker.sh
```

Error:
```
zenoh-carla-bridge does not exist. Creating...
[+] Building 0.9s (4/4) FINISHED                                                                                                                                        docker:default
 => [internal] load .dockerignore                                                                                                                                                 0.0s
 => => transferring context: 2B                                                                                                                                                   0.0s
 => [internal] load build definition from Dockerfile_carla_bridge                                                                                                                 0.0s
 => => transferring dockerfile: 927B                                                                                                                                              0.0s
 => ERROR [internal] load metadata for ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64                                                         0.9s
 => [auth] autowarefoundation/autoware-universe:pull token for ghcr.io                                                                                                            0.0s
------
 > [internal] load metadata for ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64:
------
Dockerfile_carla_bridge:1
--------------------
   1 | >>> FROM ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64
   2 |     # TODO: Don't upgrade it until we bump the carla-sys version.
   3 |     
--------------------
ERROR: failed to solve: ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64: pulling from host ghcr.io failed with status code [manifests galactic-20221201-prebuilt-cuda-amd64]: 403 Forbidden
/usr/lib/python3/dist-packages/paramiko/transport.py:237: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
Extension volume doesn't support default arguments. Please extend it.
Active extensions ['network', 'nvidia', 'privileged', 'user', 'volume', 'x11']
Step 1/6 : FROM golang:1.19 as detector
 ---> 80b76a6c918c
Step 2/6 : RUN git clone -q https://github.com/dekobon/distro-detect.git &&     cd distro-detect &&     git checkout -q 5f5b9c724b9d9a117732d2a4292e6288905734e1 &&     CGO_ENABLED=0 go build .
 ---> Using cache
 ---> 7dbb148a439f
Step 3/6 : FROM zenoh-carla-bridge
no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'zenoh-carla-bridge', maybe the base image does not exist
```

May I know to how fix it or I am missing something? Thank you.

evshary, 2024-03-11:
I guess there are some issues on `ghcr.io` right now.
Maybe you can try later.
BTW, is there any reason why you want to use galactic version?

evshary, 2024-03-12:
It seems like `ghcr.io` is back. It should not be a problem now.

ChinYikMing, 2024-03-12:
> It seems like `ghcr.io` is back. It should not be a problem now.

Yes, I will close this issue. Thank you!

closed 2024-03-12 by ChinYikMing