evshary/autoware_carla_launch #65: Failed to execute run-bridge-docker.sh

TianYifan-Vincent, 2024-03-14:
`./container/run-bridge-docker.sh

zenoh-carla-bridge does not exist. Creating...
[+] Building 3.1s (6/11)                                                        docker:default
 => [internal] load build definition from Dockerfile_carla_bridge                         0.0s
 => => transferring dockerfile: 927B                                                      0.0s
 => [internal] load metadata for ghcr.io/autowarefoundation/autoware-universe:galactic-2  2.8s
 => [internal] load .dockerignore                                                         0.0s
 => => transferring context: 2B                                                           0.0s
 => [1/8] FROM ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-c  0.0s
 => CACHED [2/8] RUN apt-get update                                                       0.0s
 => ERROR [3/8] RUN apt-get install -y wget unzip tmux inetutils-ping curl vim            0.2s
------                                                                                         
 > [3/8] RUN apt-get install -y wget unzip tmux inetutils-ping curl vim:
0.205 Reading package lists...
0.218 Building dependency tree...
0.220 Reading state information...
0.222 E: Unable to locate package tmux
0.222 E: Unable to locate package inetutils-ping
0.222 E: Unable to locate package vim
------
Dockerfile_carla_bridge:7
--------------------
   5 |     RUN apt-get update
   6 |     # Useful tools
   7 | >>> RUN apt-get install -y wget unzip tmux inetutils-ping curl vim
   8 |     RUN python3 -m pip install gdown
   9 |     RUN printf 'set-option -g default-shell /bin/bash\nset -g default-terminal "tmux-256color"' > /etc/tmux.conf
--------------------
ERROR: failed to solve: process "/bin/bash -o pipefail -c apt-get install -y wget unzip tmux inetutils-ping curl vim" did not complete successfully: exit code: 100
Extension volume doesn't support default arguments. Please extend it.
Active extensions ['network', 'nvidia', 'privileged', 'user', 'volume', 'x11']
Step 1/6 : FROM golang:1.19 as detector
 ---> 80b76a6c918c
Step 2/6 : RUN git clone -q https://github.com/dekobon/distro-detect.git &&     cd distro-detect &&     git checkout -q 5f5b9c724b9d9a117732d2a4292e6288905734e1 &&     CGO_ENABLED=0 go build .
 ---> Using cache
 ---> 3451c367aa35
Step 3/6 : FROM zenoh-carla-bridge
no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'zenoh-carla-bridge', maybe the base image does not exist`

Does anyone have any experience with this bug? I have tried the solution in the issue related to rocker.

SunBK201, 2024-03-15:
same problem

evshary, 2024-03-15:
I think it's because `apt-get update` doesn't run, so the packages can't be found. Could you clean up the cache and retry again?

SunBK201, 2024-03-15:
Is there something wrong with this autoware-universe:galactic-20221201-prebuilt-cuda-amd64 image?

evshary, 2024-03-16:
The problem is here
```
 => CACHED [2/8] RUN apt-get update                                                       0.0s
 => ERROR [3/8] RUN apt-get install -y wget unzip tmux inetutils-ping curl vim            0.2s
```

Errors shouldn't occur while trying to `apt-get install`, so I think the problem comes from it didn't do `apt-get update` correctly.

TianYifan-Vincent, 2024-03-16:
I reinstalled a lower version of docker from the official website and the problem was solved.

closed 2024-03-16 by TianYifan-Vincent