evshary/autoware_carla_launch #75: GILBC not found when run Python Agent (In Carla bridge container)

TianYifan-Vincent, 2024-03-24:
I met some issues "GILBC not found" when run Python Agent (In Carla bridge container).
./script/run-bridge.sh
```
/home/dell/autoware_carla_launch/poetry/venv/bin/python: /usr/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /home/dell/autoware_carla_launch/poetry/venv/bin/python)
/home/dell/autoware_carla_launch/poetry/venv/bin/python: /usr/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/dell/autoware_carla_launch/poetry/venv/bin/python)
/home/dell/autoware_carla_launch/poetry/venv/bin/python: /usr/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/dell/autoware_carla_launch/poetry/venv/bin/python)
/home/dell/autoware_carla_launch/poetry/venv/bin/python: /usr/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/dell/autoware_carla_launch/poetry/venv/bin/python)
```
I am using the main branch. Ubuntu 22.04 Carla 0.9.14. This docker image was created all at once, how should I go about fixing this? Should I Change to the humble branch and retry?

TianYifan-Vincent, 2024-03-26:
@evshary Sorry to bother you, have you ever had a similar problem?

evshary, 2024-04-02:
No, and I cannot reproduce your issue.
Could you clean the environment (`make clean`) and retry again?
If the issue still exists, maybe you need to remove the docker image.

TianYifan-Vincent, 2024-04-03:
@RogerJung Yeah, thank you so much! You are so kind, it works! But it may lead to some problems about python environment. Do I need to do something else?
```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = '/autoware/install/velodyne_msgs/lib/python3.8/site-packages:/autoware/install/static_centerline_optimizer/lib/python3.8/site-packages:/autoware/install/operation_mode_transition_manager/lib/python3.8/site-packages:/autoware/install/control_performance_analysis/lib/python3.8/site-packages:/autoware/install/tier4_vehicle_msgs/lib/python3.8/site-packages:/autoware/install/tier4_v2x_msgs/lib/python3.8/site-packages:/autoware/install/tier4_api_msgs/lib/python3.8/site-packages:/autoware/install/tier4_system_msgs/lib/python3.8/site-packages:/autoware/install/tier4_simulation_msgs/lib/python3.8/site-packages:/autoware/install/tier4_rtc_msgs/lib/python3.8/site-packages:/autoware/install/planning_debug_tools/lib/python3.8/site-packages:/autoware/install/tier4_planning_msgs/lib/python3.8/site-packages:/autoware/install/dummy_perception_publisher/lib/python3.8/site-packages:/autoware/install/tier4_perception_msgs/lib/python3.8/site-packages:/autoware/install/tier4_localization_msgs/lib/python3.8/site-packages:/autoware/install/tier4_hmi_msgs/lib/python3.8/site-packages:/autoware/install/autoware_external_api_msgs/lib/python3.8/site-packages:/autoware/install/tier4_external_api_msgs/lib/python3.8/site-packages:/autoware/install/tier4_debug_msgs/lib/python3.8/site-packages:/autoware/install/tier4_control_msgs/lib/python3.8/site-packages:/autoware/install/simulator_compatibility_test/lib/python3.8/site-packages:/autoware/install/morai_msgs/lib/python3.8/site-packages:/autoware/install/grid_map_msgs/lib/python3.8/site-packages:/autoware/install/autoware_sensing_msgs/lib/python3.8/site-packages:/autoware/install/autoware_planning_msgs/lib/python3.8/site-packages:/autoware/install/autoware_map_msgs/lib/python3.8/site-packages:/autoware/install/autoware_localization_msgs/lib/python3.8/site-packages:/autoware/install/autoware_control_msgs/lib/python3.8/site-packages:/autoware/install/autoware_common_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_vehicle_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_system_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_planning_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_perception_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_mapping_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_geometry_msgs/lib/python3.8/site-packages:/autoware/install/autoware_auto_control_msgs/lib/python3.8/site-packages:/autoware/install/autoware_adapi_version_msgs/lib/python3.8/site-packages:/autoware/install/autoware_adapi_v1_msgs/lib/python3.8/site-packages:/opt/ros/galactic/lib/python3.8/site-packages'
  program name = '/home/dell/autoware_carla_launch/poetry/venv/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/home/dell/autoware_carla_launch/poetry/venv/bin/python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.platlibdir = 'lib'
  sys.executable = '/home/dell/autoware_carla_launch/poetry/venv/bin/python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/autoware/install/velodyne_msgs/lib/python3.8/site-packages',
    '/autoware/install/static_centerline_optimizer/lib/python3.8/site-packages',
    '/autoware/install/operation_mode_transition_manager/lib/python3.8/site-packages',
    '/autoware/install/control_performance_analysis/lib/python3.8/site-packages',
    '/autoware/install/tier4_vehicle_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_v2x_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_api_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_system_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_simulation_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_rtc_msgs/lib/python3.8/site-packages',
    '/autoware/install/planning_debug_tools/lib/python3.8/site-packages',
    '/autoware/install/tier4_planning_msgs/lib/python3.8/site-packages',
    '/autoware/install/dummy_perception_publisher/lib/python3.8/site-packages',
    '/autoware/install/tier4_perception_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_localization_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_hmi_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_external_api_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_external_api_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_debug_msgs/lib/python3.8/site-packages',
    '/autoware/install/tier4_control_msgs/lib/python3.8/site-packages',
    '/autoware/install/simulator_compatibility_test/lib/python3.8/site-packages',
    '/autoware/install/morai_msgs/lib/python3.8/site-packages',
    '/autoware/install/grid_map_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_sensing_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_planning_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_map_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_localization_msgs/lib/python3.8/site-packages',
[... cut ...]
    '/autoware/install/autoware_auto_control_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_adapi_version_msgs/lib/python3.8/site-packages',
    '/autoware/install/autoware_adapi_v1_msgs/lib/python3.8/site-packages',
    '/opt/ros/galactic/lib/python3.8/site-packages',
    '/usr/lib/python310.zip',
    '/usr/lib/python3.10',
    '/usr/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'
```

evshary, 2024-05-05:
Hi @TianYifan-Vincent 
I've updated the Ubuntu version in the bridge container. Maybe you can try and see whether the issue still exists or not.

evshary, 2024-05-27:
I'll close the issue and feel free to reopen it if you have any further questions.

closed 2024-05-27 by evshary