evshary/autoware_carla_launch #82: How to detect the traffic light in Carla?

TianYifan-Vincent, 2024-04-23:
Sorry to bother you @evshary , i want to know whether in this bridge, Autoware can recognize the traffic light in Carla and follow the traffic rules.

evshary, 2024-04-26:
Hi @TianYifan-Vincent, the bridge only passes the camera image to Autoware. We still need to do some configuration in Autoware to be able to recognize the traffic light.

TianYifan-Vincent, 2024-04-26:
> Hi @TianYifan-Vincent, the bridge only passes the camera image to Autoware. We still need to do some configuration in Autoware to be able to recognize the traffic light.

Thank you so much and your devotion!

closed 2024-04-26 by TianYifan-Vincent

Justin-Xiang, 2025-01-27:
@evshary Hi, Could you provide guidance on configuring Autoware for traffic light recognition? I came across [this section](https://autowarefoundation.github.io/autoware.universe/main/simulator/autoware_carla_interface/#traffic-light-recognition), which mentions modifying the map. Is there a more convenient or accurate method to achieve this? Any advice or best practices would be greatly appreciated!

Thank you!

evshary, 2025-02-03:
Hi @Justin-Xiang I'm afraid there are no other convenient ways to do so since we still need the traffic light information on the map.