evshary/autoware_carla_launch #83: supported features

Leason-ly, 2024-04-26:
hello， @evshary ，I have successfully replicated multiple vehicles in parallel according to Readme. I would like to ask if this architecture supports interaction between multiple vehicles, such as perceiving other vehicles and avoiding them, identifying and following traffic signals? 
Will future versions integrate these features? If it's not supported, how can I expand it？

evshary, 2024-05-05:
Hi @Leason-ly 
Yes, this is on the roadmap, but I'm not sure when this will be implemented. (busy on other projects now)
The concept is we want to limit the ROS 2 (DDS) traffic inside the Autoware, and use Zenoh to communicate with outside entities.
We've had a simple scenario for V2X in the repository. Maybe you want to take a look.
https://github.com/evshary/zenoh_autoware_v2x

evshary, 2024-05-27:
I'll close the issue and feel free to reopen it if you have any further questions.

closed 2024-05-27 by evshary