evshary/autoware_carla_launch #85: Carla port

JessicaFerrara, 2024-05-01:
hello. I would like to ask if I changed the Carla port (I don't use the default 2000) when I run the Carla server how do I change it also when I run the file in the Carla bridge container?

epris, 2024-05-03:
Hello,

just to clarify the problem:
- we changed the Carla server port to 3000. So we write "./CarlaUE4.sh -quality-level=Epic -world-port=3000 -RenderOffScreen"
- then, we modified "main.py" so that args.port is equal to **3000**
- we obtained the following error 
"terminate called after throwing an instance of 'carla::client::TimeoutException'
  what():  time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to [127.0.0.1:2000](http://127.0.0.1:**2000**/)"
- We are sure that in simulation.loop.py client = carla.Client(args.host, args.port) args.port is equal to 3000
- We tried to launch the script simple_spawn.py and the connection with the server happened and we got the following message "Hello from the pygame community. https://www.pygame.org/contribute.html
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 922 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 923 (sensor.other.imu)"

We are not able to understand why a timeout on port 2000 is still notified if we set the port to 3000

evshary, 2024-05-05:
Hi @JessicaFerrara and @epris 
You only changed the port in Python script but not the bridge.
Please add `--carla-port 3000` in the script here
https://github.com/evshary/autoware_carla_launch/blob/main/script/run-bridge.sh#L12

epris, 2024-05-20:
> Hi @JessicaFerrara and @epris
> You only changed the port in Python script but not the bridge.
> Please add `--carla-port 3000` in the script here
> https://github.com/evshary/autoware_carla_launch/blob/main/script/run-bridge.sh#L12

it works, thanks!

closed 2024-05-27 by evshary