evshary/autoware_carla_launch #95: Cannot edit Town01.osm map

xiaoc88, 2024-05-27:
map tool: https://tools.tier4.jp/vector_map_builder_ll2/
osm map: Town01.osm
error: FileFormatError,Map exists over multiple grid-zones

[image]

evshary, 2024-05-28:
Hi @xiaoc88, in fact, the map we're using now is from other project
https://github.com/evshary/autoware_carla_launch/tree/main/carla_map
I didn't do too much research on how to edit the map. I'm afraid I can't give you a proper answer.

xiaoc88, 2024-05-28:
@evshary Thank you for your answer, should I close this question now?

evshary, 2024-05-28:
Maybe you can close it. But still feel free to put any other information here if you find the solution. It would be a good reference for others.

xiaoc88, 2024-05-28:
Okay, I will close this issue first.

closed 2024-05-28 by xiaoc88