evshary/autoware_carla_launch #96: The ego-vehicle in Autoware is not in the list of CARLA vehicles

epris, 2024-06-11:
Dear all,

we have successfully used the bridge between Autoware and Carla but we have noticed the following issue that we are not able to solve it:
In Autoware, the ego-vehicle receives data from camera and lidar sensors from CARLA but this vehicle doesn't appear in the list of CARLA's vehicles. It's like the sensors are attached to a phantom vehicle. If we spawn an ego-vehicle in CARLA it will be different from the ego-vehicle in Autoware.
Is there are way to overcome this issue? Did we use the bridge in the wrong way?

Thanks a lot,
Antonio

evshary, 2024-06-12:
Hi @epris 

I'm a little unclear about what the phantom vehicle means. Maybe you can have more explanation of how you list vehicles?
BTW, here is how we create the car. I would suggest that you could check the code and see whether it matches your expectations or not.
https://github.com/evshary/zenoh_carla_bridge/blob/b9ad9606a3dc1e5310fe5a926a4b23c3c3a5e11c/carla_agent/simulation/world.py#L95

evshary, 2024-09-27:
I'll close the issue. Feel free to reopen it if there is any further problem.

closed 2024-09-27 by evshary