evshary/autoware_carla_launch #98: rosbag2 recording for Autoware's topic

TangChenTom, 2024-06-15:
Dear developer,
     Is there any method to use rosbag2 or other methods to record the topics and nodes of Autoware.Universe? I have tried to go into the container for Carla bridge and container for Zenoh+Autoware, but the ros2 topic list only return to parameter and rosout. 
     My environment is Ubuntu 22.04 with Carla 0.9.14. Is this a bug I met or I tried to record the topic and node by a wrong method?
     
Sincerely Thanks!
Tom

evshary, 2024-06-17:
Hi Tom, you could take a look at [the issue](https://github.com/evshary/autoware_carla_launch/issues/25). Maybe it can solve your problem.

TangChenTom, 2024-06-17:
Thanks for your reply! I would check the issue you mentioned. Thanks for your work on this project!

evshary, 2024-09-27:
I'll close the issue. Feel free to reopen it if there is any further problem.

closed 2024-09-27 by evshary