evshary/autoware_carla_launch #103: failed to run custom build command for `carla-sys v0.11.1`

xiaoc88, 2024-07-01:
Hi, I ran into an issue where Cara-sys would not build.
command: make build_bridge
error:
[image]

Or always in such a state:
[image]

The following methods have been tried but still cannot be solved:
1. Replace Cargo.toml with the link content
https://github.com/evshary/zenoh_carla_bridge/blob/main/Cargo.lock#L853
2. Delete Cargo.lock
3. https://github.com/evshary/autoware_carla_launch/issues/52

evshary, 2024-07-02:
Hi @xiaoc88 It seems like you can't download the artifact from GitHub. Could you try to ping / wget the data from GitHub?

xiaoc88, 2024-07-02:
@evshary Is my side of the problem,Clean the carla bridge container，and then compile a few more times can be, thank you.

closed 2024-07-02 by xiaoc88