evshary/autoware_carla_launch #108: Potential ros2dds config issue after 0.11.0 update

MamoruDS, 2024-07-04:
Hello and thanks for maintaining this repository.

The current configuration for ros2dds is outdated due to changes in zenoh-plugin-ros2dds version 0.11.0. Previously, as in the `humble` branch, ros2dds operated in **peer** mode. However, it now defaults to **router** mode on [this change](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/119). This leads to potential port conflicts with `zenoh_carla_bridge` when both are run in containerized environments on the same network interface without specified listening ports. 

I am prepared to submit a PR to update the ros2dds config and align it with the `humble` branch setup.

evshary, 2024-07-05:
Hi @MamoruDS, Thanks for your contribution! Is there any reason why you run both containers on the same network interface? By default, we use the host interface for the bridge container, but not the autoware container. Or do you mean you run both bridge and autoware in the same container?

MamoruDS, 2024-07-05:
Thanks for the reply! For remote access, we need to use the `host` network to enable Xforwarding over ssh, which serves as a workaround for xauth. As a result, both containers use the `host` network, leading to a port conflict since both zenoh instances run in router mode and listen on `0.0.0.0:7447`.
Although this configuration is somewhat specific to our setup, it worked smoothly with the older ros2dds version. It would be helpful to retain this behavior in `humble-0.9.14`.

BTW, either `client` or `peer` mode works, and from the log, it seems the `ros2dds` [version](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/tree/78843980de77fa9718d2a75425c423e6d604a8f3) in the `humble` branch was running in `peer` mode. For future deployments involving bridged Autoware ROS2 interfaces and utilizing the zenoh P2P network, `peer` mode seems preferable :)

evshary, 2024-07-05:
TBH, I'm thinking of changing the default port of `zenoh_carla_bridge` to avoid port conflict... I prefer to use the default mode of `ros2dds` because it will be the same behavior in `rmw_zenoh` in the future.

MamoruDS, 2024-07-05:
That sounds fair to me. Thank you for your response!

closed 2024-07-05 by MamoruDS