evshary/autoware_carla_launch #113: how to generate steer map for autoware to control carla vehicle

ahsan155, 2024-07-07:
I am not using the bridge you built. I am currently trying to generate steer map that raw_vehicle_cmd_converter in autoware uses to send accurate steering command to carla vehicle. Did you generate this steer_map.csv file. If so, what calibrator did you use. I am trying this ml based calibrator [https://github.com/pixmoving-moveit/learning_based_vehicle_calibration/tree/normal_vehicle](url). autoware documentation doesn't say how to generate the steer map file.

I would really appreciate if you can give me some pointers on how to generate steer map file or any alternative ways to correctly send steering command.

evshary, 2024-07-09:
I'm afraid I can't help you with this. I know little about the steer map. Maybe you can ask the question in the Autoware discussion and see if there is anyone who knows more about this topic.

evshary, 2024-09-27:
I'll close the issue. Feel free to reopen it if there is any further problem.

closed 2024-09-27 by evshary