evshary/autoware_carla_launch #114: Add sensor “lidar.ray_cast_semantic”

JessicaFerrara, 2024-07-08:
Hi. 
I need to add the sensor "sensor.lidar.ray_cast_semantic" in Carla, I have a script where I spawn the sensor. But when I run the bridge it gives the error "CarlaIssue: 'sensor not exists'". I don't need to use it in autoware so I don't need to pass any data to autoware, just into Carla. I also tried editing sensor bridge files like world.py or lidar.py but the error did not resolve. To add this sensor do I need to edit any particular files or do anything else?

evshary, 2024-07-09:
I think there is a bug here
https://github.com/evshary/zenoh_carla_bridge/blob/ea2c19861caeded9ff44109f2403c6b955975e96/src/autoware.rs#L134
This should be modified as `list_lidar_semantics`.

However, you can just comment the `SensorType::LidarRayCastSemantic` part and compile again if you don't need to bridge this to the Autoware.

evshary, 2024-09-27:
I'll close the issue. Feel free to reopen it if there is any further problem.

closed 2024-09-27 by evshary