evshary/autoware_carla_launch #124: Not able to initialize the autoware agent

PeterMal0ne, 2024-08-03:
Hello! Thanks for actively maintaining this repo. 

I followed all the steps in the documentation, but when the autoware was launched, I got "no-image" on the default "/perception/traffic_light_recognition/camera6/debug/rois" topic. And it seems to be same empty topic for others. 

On the side of terminal output, I kept getting this error: [ERROR] [1722718763.870537177] [system.service_log_checker]: /localization/initialize: status code 1 'The vehicle is not stopped.' (/localization/util/pose_initializer_node) while I am trying to set the initial pose for the vehicle. 

Here is the full output of one round of "pose initialization": 

[service_log_checker-4] [ERROR] [1722718796.722090621] [system.service_log_checker]: /localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/node/localization)
[service_log_checker-4] [ERROR] [1722718796.722133374] [system.service_log_checker]: /api/localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/node/localization)
[component_container-64] [INFO] [1722718797.529683160] [planning.scenario_planning.parking.costmap_generator]: waiting for initial pose...
[initial_pose_adaptor-78] [INFO] [1722718797.721587826] [default_ad_api.helpers.initial_pose_adaptor]: waiting response
[initial_pose_adaptor-78] [INFO] [1722718797.927563001] [default_ad_api.helpers.initial_pose_adaptor]: Loaded partial pcd map from map_loader (grid size: 1)
[initial_pose_adaptor-78] [INFO] [1722718798.158808398] [default_ad_api.helpers.initial_pose_adaptor]: modified point: 10.000 20.000 -3.144
[initial_pose_adaptor-78] [INFO] [1722718798.158886730] [default_ad_api.helpers.initial_pose_adaptor]: client call: /api/localization/initialize
[initial_pose_adaptor-78] [INFO] [1722718798.159030022] [default_ad_api.helpers.initial_pose_adaptor]: original point: 10.000 20.000 0.000
[initial_pose_adaptor-78] [INFO] [1722718798.159429069] [default_ad_api.helpers.initial_pose_adaptor]: Send request to map_loader
[component_container_mt-76] [INFO] [1722718798.159105321] [default_ad_api.node.localization]: server call: /api/localization/initialize
[component_container_mt-76] [INFO] [1722718798.159703233] [default_ad_api.node.localization]: client call: /localization/initialize
[pose_initializer_node-31] [INFO] [1722718798.159830340] [localization.util.pose_initializer_node]: server call: /localization/initialize
[pose_initializer_node-31] [INFO] [1722718798.159901283] [localization.util.pose_initializer_node]: server exit: /localization/initialize

Any help is appreciated. Thank you in advance. 

[image]

evshary, 2024-08-06:
Hi @PeterMal0ne 
Could you please provide the information on what branch you are using?
Also providing the logfile will be helpful. It should be under the bridge_log and autoware_log folder.

PeterMal0ne, 2024-08-18:
Thanks for the replying! 

I am using the main branch, and here is my log file. In addition to what I posted before, I noticed the topics are all up in the **_autoware_** container, but there is nothing as I showed in the _**bridge**_ container.

Here is the bridge.log:
 INFO  zenoh_carla_bridge > Running Carla Autoware Zenoh bridge...
 INFO  zenoh::net::runtime > Using ZID: c542051a5fe7c408f8b0c09f9e275323
 INFO  zenoh::net::runtime::orchestrator > Zenoh can be reached at: tcp/192.168.50.156:7447
 INFO  zenoh::net::runtime::orchestrator > Zenoh can be reached at: tcp/172.17.0.1:7447
 INFO  zenoh::net::runtime::orchestrator > zenohd listening scout messages on 224.0.0.224:7446

 INFO  zenoh_carla_bridge::bridge::sensor_bridge > Detected a sensor 'godview' on 'v1'
 INFO  zenoh_carla_bridge                        > Actor 204 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge > Detected a sensor 'tamagawa' on 'v1'
 INFO  zenoh_carla_bridge                        > Actor 201 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge > Detected a sensor 'traffic_light' on 'v1'
 INFO  zenoh_carla_bridge                        > Actor 203 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge > Detected a sensor 'top' on 'v1'
 INFO  zenoh_carla_bridge                        > Actor 202 created
 INFO  zenoh_carla_bridge::bridge::vehicle_bridge > Detect a vehicle v1
 INFO  zenoh_carla_bridge                         > Actor 197 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge  > Detected a sensor 'collision' on 'v1'
 WARN  zenoh_carla_bridge::bridge::sensor_bridge  > Collision sensor is not supported yet
 INFO  zenoh_carla_bridge                         > Actor 198 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge  > Detected a sensor 'ublox' on 'v1'
 INFO  zenoh_carla_bridge                         > Actor 200 created
 INFO  zenoh_carla_bridge::bridge::sensor_bridge  > Detected a sensor 'lane_invasion' on 'v1'
 WARN  zenoh_carla_bridge::bridge::sensor_bridge  > Unsupported sensor type 'sensor.other.lane_invasion'
 INFO  zenoh_carla_bridge                         > Actor 199 created
 
and vehicle.log:
INFO: listening to server 127.0.0.1:2000
ALSA lib confmisc.c:855:(parse_card) cannot find card '0'
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_card_inum returned error: No such file or directory
ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM default

Where I think the ALSA lib should not cause the issue. I also attached the autoware log. 
[launch.log](https://github.com/user-attachments/files/16647076/launch.log)

(I also have tried the humble branch as it is mentioned more stable in README, but seems autoware is not providing the elder docker version, specifically, this one:  ghcr.io/autowarefoundation/autoware-universe:galactic-20221201-prebuilt-cuda-amd64)
 
Please let me know if you need more information!

habby1012, 2024-08-23:
Hi, @PeterMal0ne!,
The information shown in your bridge.log from the bridge container looks correct and sufficient. However, you mentioned that the topics are all up in the Autoware container. Are they successfully displayed in RViz, such as the camera view, localization, and so on?

PeterMal0ne, 2024-08-29:
> Hi, @PeterMal0ne!, The information shown in your bridge.log from the bridge container looks correct and sufficient. However, you mentioned that the topics are all up in the Autoware container. Are they successfully displayed in RViz, such as the camera view, localization, and so on?

@habby1012 Thanks for the reply. No, they are not displayed in RViz, and that is the part that confused me. I saw the car spawned in the CARLA but none of the information is displayed in the RViz. And I got this in my Auotware container:

1723950013.3359039 [component_container-64] [INFO] [1723950013.335673204] [planning.scenario_planning.parking.costmap_generator]: waiting for initial pose...

When I am trying to set the initial pose using RViz, it did not set the position of the vehicle.

evshary, 2024-09-09:
Hi @PeterMal0ne 
Maybe you can try the main branch now. I fixed the docker image issue there.
Also, I didn't see the log of `zenoh_bridge_ros2dds.log`. Does it run successfully?

evshary, 2024-11-28:
I will close the issue. If you have any further questions, feel free to reopen it.

closed 2024-11-28 by evshary