evshary/autoware_carla_launch #125: Unable to run scripts associated to the "run-bridge-docker" container

KethanR, 2024-08-05:
Hi ChenYing (CY),

I successfully built the new/updated docker containers (as of 5th August), but I am now unable to run any scripts related to the "run-bridge-docker" container. For example, when I attempt to run the following:
[image]
I get hit with the following error:
[image]

Do you know what the issue could be, and how you'd go about fixing it?  
Thank you for actively maintaining the repository, and for all the help in advance.  

Warm regards,
Kethan

**Disclaimer: I am currently working off the main branch.**

evshary, 2024-08-06:
Hi @KethanR 
From your error log, it seems like the bridge didn't build successfully.

We are also working on updating the Autoware version
https://github.com/evshary/autoware_carla_launch/pull/101
We didn't face the same issues, but it took a longer time to do the localization. We are still investigating. Maybe you can give it a try.

evshary, 2024-11-28:
I will close the issue. If you have any further questions, feel free to reopen it.
BTW, the Autoware is updated to the September version.

closed 2024-11-28 by evshary