evshary/autoware_carla_launch #129: How to use Traffic Manager with the Bridge

epris, 2024-08-29:
Hello,

I would like to configure the Traffic Manager (TM) so that it works in deterministic mode. In deterministic mode, the TM will produce the same results and behaviors under the same conditions.
Is that possible with the bridge?

We tried the code below but the inclusion of the instruction related to the TM blocks the non-player characters.
Thanks a lot

`import carla
import random
import time

# Connect to the client and retrieve the world object
client = carla.Client('localhost', 3000)
world = client.get_world()
#world = client.load_world('Town03')

If we configure the TM as in the code below the simulation doesn't start
# Set up the simulator in synchronous mode
settings = world.get_settings()
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.05
world.apply_settings(settings)

# Set up the TM in synchronous mode
traffic_manager = client.get_trafficmanager()
traffic_manager.set_synchronous_mode(True)

# Set a seed so behaviour can be repeated if necessary
traffic_manager.set_random_device_seed(0)
random.seed(0)

# We will aslo set up the spectator so we can see what we do
spectator = world.get_spectator()

spawn_points = world.get_map().get_spawn_points()`

evshary, 2024-09-09:
Hi @epris I didn't try TM before, but we are using sync mode by default.
Could you provide some information, like which branch you are using now and where you add these code (in our Python script here? https://github.com/evshary/autoware_carla_launch/blob/1ee137b61dc37df0547a56ac9ee89fe842ba9033/script/run-bridge.sh#L17)

JessicaFerrara, 2024-09-17:
Hello
we set the bridge in sync mode in the file run_bridge.sh and now the Traffic Manager works. The code above is a piece of a script that creates a scenario with the spawn of several players. The problem is that by setting the seed (both random seed and Traffic Manager seed as in the code above) and repeating the same scenario several times, the behavior is not the same, in particular the behavior of the ego-vehicle is different for each run (the velocity of vehicle is different) while that of the other players is the same. For both the ego-vehicle and the other players, autopilot is set.
The bridge, Carla and the Traffic Manager are in sync mode

evshary, 2024-09-24:
Hi @JessicaFerrara, if I didn't misunderstand your settings, you haven't run Autoware and just set ego-vehicle to autopilot mode, but the ego-vehicle doesn't have a consistent behavior.
Here is how I configured the ego-vehicle, perhaps you can take a look and tweak the configuration
https://github.com/evshary/zenoh_carla_bridge/blob/main/carla_agent/simulation/world.py#L83

JessicaFerrara, 2024-09-25:
Hello, I run Autoware, the bridge and the Carla server. I also run a file that creates a scenario and sets the autopilot to the ego-vehicle. In order to repeat the same scenario several times and have the same behaviour of the players, I have used the Traffic Manager with a seed, but the behaviour of the ego-vehicle is not consistent, in fact it happens that it blocks, i.e. the speed is 0 km/h, for no reason (there is no error). Below is the file i use

import time
import random
import carla

client = carla.Client('localhost', 3000)
#client.set_timeout(2.0)
world = client.get_world()
random.seed(9)
# Set up the simulator in synchronous mode
settings = world.get_settings()
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.05
world.apply_settings(settings)

# Set up the TM in synchronous mode
traffic_manager = client.get_trafficmanager()
traffic_manager.set_synchronous_mode(True)

# Set a seed so behaviour can be repeated if necessary

traffic_manager.set_random_device_seed(9)

blueprint_library = world.get_blueprint_library()

vehicle = blueprint_library.find('vehicle.tesla.model3')
vehicle2 = blueprint_library.find('vehicle.ford.mustang')
walker = blueprint_library.find('walker.pedestrian.0001')
cyclist = blueprint_library.find('vehicle.bh.crossbike')
moto = blueprint_library.find('vehicle.kawasaki.ninja')
moto2 = blueprint_library.find('vehicle.yamaha.yzf')
truck = blueprint_library.find('vehicle.carlamotors.firetruck')
truck2 = blueprint_library.find('vehicle.tesla.cybertruck')

walker_controller_bp = blueprint_library.find('controller.ai.walker')

s2=carla.Transform(carla.Location(x=55.7, y=-206.6, z=0.3), carla.Rotation(pitch=0.000000, yaw=180.00, roll=0.000000))
s3=carla.Transform(carla.Location(x=34.7, y=-196.6, z=0.3), carla.Rotation(pitch=0.000000, yaw=0.00, roll=0.000000))
s4=carla.Transform(carla.Location(x=23.8, y=-193.6, z=0.3), carla.Rotation(pitch=0.000000, yaw=0.00, roll=0.000000))
s5=carla.Transform(carla.Location(x=-28.2, y=-198.4, z=0.3), carla.Rotation(pitch=0.000000, yaw=0.00, roll=0.000000))
s6=carla.Transform(carla.Location(x=-36.6, y=-194.9, z=0.3), carla.Rotation(pitch=0.000000, yaw=0.00, roll=0.000000))
s7=carla.Transform(carla.Location(x=62.92, y=-209.88, z=0.3), carla.Rotation(pitch=0.000000, yaw=-180.00, roll=0.000000))
s8=carla.Transform(carla.Location(x=44.8, y=-212.6, z=0.3), carla.Rotation(pitch=0.000000, yaw=180.00, roll=0.000000))

v2 = world.spawn_actor(vehicle, s2)
v3 = world.spawn_actor(vehicle, s3)
v4 = world.spawn_actor(truck, s4)
v5 = world.spawn_actor(moto, s5)
v6 = world.spawn_actor(vehicle, s6)
v7 = world.spawn_actor(cyclist, s7)
v8 = world.spawn_actor(walker, s8)

controller = world.try_spawn_actor(walker_controller_bp, carla.Transform(), attach_to=v8)
walker_control = carla.WalkerControl()
walker_control.speed = 2 
walker_control.direction = carla.Vector3D(-1, 0, 0)  

time.sleep(2)

vehicle_id = 109
ego_vehicle = world.get_actor(vehicle_id)

v7.set_autopilot(True)
time.sleep(3)
v2.set_autopilot(True)
v3.set_autopilot(True)
v4.set_autopilot(True)
v5.set_autopilot(True)
v6.set_autopilot(True)
v8.apply_control(walker_control)

ego_vehicle.set_autopilot(True)

evshary, 2024-09-27:
Hi @JessicaFerrara Sorry in advance that I updated your code to make it easier to read.
I still don't quite understand what you want to achieve. Is your ego vehicle here controlled by Autoware or it's just another vehicle controlled by the traffic manager?

epris, 2024-09-27:
Hi @evshary, @JessicaFerrara and I are using the system in the following way: we run the Carla Server and the Bridge, then we run Autoware and eventually, we run the code shared by @JessicaFerrara.
The bridge spawns the ego vehicle, which is then controlled by the traffic manager with the autopilot.

evshary, 2024-09-27:
Hi @epris, I can give it a try next week and see what will happen. However, what I can't get is why the ego vehicle is controlled by both Autoware and the traffic manager. I thought it was only controlled by Autoware. Is there any reason for that?

epris, 2024-09-27:
Hi @evshary, we run Autoware because we need to record the topics published by the different nodes. In particular, we are interested in topics related to perception and tracking. When we observe the behavior of the ego vehicle in Rviz, we see that it follows the same trajectory in Carla.

evshary, 2024-10-04:
Hi @epris @JessicaFerrara could you please try the branch here?
https://github.com/evshary/zenoh_carla_bridge/tree/fix_update_value
I found that I always update the control message to Carla even when there is no message sent from Autoware. Perhaps this is why the strange behavior happens.

evshary, 2025-02-07:
I will close the issue while the fix is merged now. Feel free to open it if you have any other questions.

closed 2025-02-07 by evshary