evshary/autoware_carla_launch #133: Error during build of bridge and autoware

Georgegal00, 2024-10-28:
I'm using the branch humble-0.9.14, and ubuntu 22.04.
Following the tutorial, I ran into two problems.
1) Going to do the prepare_bridge, I was noticing how the variable $POETRY_HOME, was not being set, after source env.sh, even though I was inside a container, and so the bridge build also failed.
I think I solved this problem by changing the image in run-bridge-docker.sh: DOCKER_IMAGE=zenoh-carla-bridge-2024-09.
2) Going to do prepare_autoware I get the following message: 
“rosdep install -y --from-paths src --ignore-src --rosdistro humble.
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
carla_sensor_kit_launch: Cannot locate rosdep definition for [pointcloud_preprocessor].
autoware_manual_control: Cannot locate rosdep definition for [autoware_auto_vehicle_msgs]
carla_pointcloud: Cannot locate rosdep definition for [autoware_auto_control_msgs]
make: *** [Makefile:26: prepare_autoware] Error 1”;
Again, I noticed that the variable $POETRY_HOME is not set, but I can't find a way to fix it. Thanks for your attention and I hope you can help me.

evshary, 2024-10-29:
Hi @Georgegal00 Thanks for trying the project.
Would you mind trying the main branch? Since the docker image from Autoware has changed, we also have a corresponding update on the project. I believe the issue you faced most likely comes from the mismatched version of Autoware and the bridge.

Georgegal00, 2024-11-04:
Thank you for your response.
I'm trying with the main version and build autoware from the source code, but following the guide I stop at the command : “source autoware/install/setup.bash”, as this file is not found, which in fact does not exist in the folder. Any solution about this?

evshary, 2024-11-08:
I guess this is because you're using out-dated images. Please remove the out-dated one and try again.

Georgegal00, 2025-01-21:
Can you provide the steps to do this? I tried “git submodule update --init --recursive” and all submodules seem to point to the latest version but when I launch the ./build_autoware script I get error for ros2socketcan and transport_drivers packages during “vcs import src < autoware.repos”. I tried changing the two versions of these two packages to the “main” version. Subsequently the script manages to finish and I am able to move on. Another error I encounter during the “make prepare_autoware” , in build_models.sh concerns a file that is not found “lidar_tranfusion_param.yaml” in the required location, searching in the files throughout the repo I found one which I pasted in the required location, finishing the script well. But when I run the simulations the machine never initializes , the problem seems to be the /localization/util/pose_initializer node : “status code 4 ‘align server failed’”. This I think then leads to all the other problems and errors I get like “cannot find a connection between map and base_link ...”, “no initial_pose_provided..” , and others . I am running the simulation in a ubuntu 22.04 system , with NVIDIA-SMI 555.42.02 and cuda version: 12.5. Any information would be helpful, thank you very much.

evshary, 2025-02-03:
Could you clean the docker image & container first and then try again?
The name of the image should be `zenoh-carla-bridge-xxx` and `zenoh-autoware-xxx`.
You can use `docker system prune -a` to clean all the images (NOTE: this will also clear other docker images in your system).

If this still doesn't work, please share the log with us for further investigation.

evshary, 2025-08-05:
I will close the issue. Please feel free to open it again if you have any further questions.

closed 2025-08-05 by evshary