evshary/autoware_carla_launch #135: Vehicle stopped beyond target point

xiaoc88, 2024-11-10:
Hi @evshary 
I have encountered a problem, in the automatic driving mode, the vehicle will exceed the set target point every time it stops, how to solve this problem?

[image]

evshary, 2024-11-26:
Hi @xiaoc88 Sorry for missing your messages. This is a known issue, but we don't have time to look into it yet.

Stuare8, 2025-01-06:
@evshary Hello! Thanks for actively maintaining this project.
I also had this problem when the vehicle would still go forward some distance after reaching the target point,I did not find this problem using planning simulation.
autoware_carla_launch branch:main
carla version: 0.9.14
autoware version: 2024.09
I compared the longitudinal speed of `/control/command/control_cmd` and `/vehicle/status/velocity_status` in both carla simulation and the planning simulator.
planning simulator:
During acceleration and deceleration, the vehicle's speed closely tracks the control speed.

https://github.com/user-attachments/assets/f7b04769-a9f3-48b1-8234-e5351732c53f

carla simulation:
During acceleration and deceleration, the vehicle speed exhibits a lag relative to the control speed.

https://github.com/user-attachments/assets/d6b3700e-e5d6-47c0-a198-d2a39757202a

Do you have any suggestions or solutions for this issue?  
Thank you so much for your answers.

evshary, 2025-02-07:
@Stuare8 Thank you for the clear video. We actually found out what might have caused the issue and are debugging it.

hsule, 2025-02-17:
Hello @Stuare8, we have fix this issue in [PR #153 – Switch vehicle control from Ackermann to actuation](https://github.com/evshary/autoware_carla_launch/pull/153), which has now been merged into the main branch.  
The problem was caused by a mismatch between Autoware's speed logic and CARLA's motion constraints. We switched the vehicle control method from Ackermann commands (speed and accelaration control) to Actuation commands (direct throttle and brake control).  
Please update to the latest commit in the main branch to fix the issue, thank you!

evshary, 2025-02-27:
Fixed via #153

closed 2025-02-27 by evshary