evshary/autoware_carla_launch #140: how can i use my lanelet map and carla map

zsdfaker, 2024-11-26:
I constructed a Carla map by using an OpenDrive map and an FBX model, and it functioned properly in Carla. I generated a Lanelet map and a PCD based on the OpenDrive file. However, when I set CARLA_MAP_NAME as my map, follow error occurred.

```
[rviz2-80] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-80]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp
[rviz2-80] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-80]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp
[rviz2-80] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```

My workpace is ubuntu 20.04 ros2 galactic carla 0.9.13

zsdfaker, 2024-11-26:
I successfully loaded the self-built map and managed to spawn vehicles in Carla, and also obtained the data from the on-board cameras. However, the vehicles failed to be spawned successfully in Autoware.
[image]

evshary, 2024-11-28:
I didn't try to construct the customized map before, so might not be able to provide some valuable comments.
Perhaps you could give an initial pose in rviz and see whether it helps or not. I guess something went wrong in GNSS and it can't be localized successfully.

zsdfaker, 2024-11-28:
When I set it manually

```
ros2 topic pub /initialpose geometry_msgs/msg/PoseWithCovarianceStamped "{
  header: {
    frame_id: 'map'
  },
  pose: {
    pose: {
      position: {
        x: 93.85017395019531,
        y: -79.0370101928711,
        z: 0.0
      },
      orientation: {
        x: 0.0000,
        y: 0.0000,
        z: 0.9639,
        w: 0.2661
      }
    }
  }
}" --once
```

the rotation of the car is incorrect.
[image]