evshary/autoware_carla_launch #148: Would Autoware without CUDA be supported?

Justin-Xiang, 2025-01-23:
I'm trying to run this project on a machine without CUDA. To achieve this, I made the following changes:
1. Updated the base Docker image from `FROM ghcr.io/autowarefoundation/autoware:20240903-devel-cuda-amd64` to `FROM ghcr.io/autowarefoundation/autoware:20240903-devel-amd64`. However, I'm not entirely sure if this change is sufficient.
2. Removed the `--nvidia` flag from the container start script, as the machine does not have an NVIDIA GPU.

These changes allowed me to successfully build the `bridge` container. However, when attempting to build the container for Zenoh + Autoware using `make prepare_autoware`, I encountered the following error. It appears to be CUDA-related, as `autoware_lidar_centerpoint` requires CUDA to run:

```
# Prebuild models
./script/build_models.sh
[INFO] [launch]: All log files can be found below /home/$USER/.ros/log/xxx
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): package 'autoware_lidar_centerpoint' found at '/opt/autoware', but libexec directory '/opt/autoware/lib/autoware_lidar_centerpoint' does not exist
make: *** [Makefile:29: prepare_autoware] Error 1
```
Any suggestions or comments would be welcomed!

evshary, 2025-02-03:
Hi @Justin-Xiang I'm not pretty sure whether we can run Autoware in Carla simulator without using CUDA. However, you can remove this line to avoid downloading models while running `make prepare_autoware`. I don't think downloading models is required if you run Autoware without CUDA.
https://github.com/evshary/autoware_carla_launch/blob/a452cf83e1b93a1ad679f8bae4f02348ba2b86fc/Makefile#L24

Justin-Xiang, 2025-02-03:
Hi @evshary, thanks for your response! However, is the issue really related to downloading models? The error seems to occur during the build process rather than when fetching the models. I suspect that replacing the lidar module might be a potential workaround, but that would require significant effort.

That said, I won’t be pursuing this further since I no longer need to run Autoware on a machine without a GPU. Feel free to reopen the issue if you encounter the same problem in the future!

closed 2025-02-03 by Justin-Xiang

evshary, 2025-02-06:
> Hi [@evshary](https://github.com/evshary), thanks for your response! However, is the issue really related to downloading models? The error seems to occur during the build process rather than when [...]

We run some ROS node not only downloading models but also doing some transformation. That's why you see the error.