evshary/autoware_carla_launch #156: Image in RViz freezes or fails to display when starting Autoware

habby1012, 2025-02-28:
**Issue Description**
I am encountering frequent issues with image display in RViz when launching Autoware. The image sometimes freezes or does not show up.

**Steps to Reproduce**

Case 1:
- Start Carla bridge and Autoware.
- Observe that the image in RViz sometimes gets stuck or displays "No Image."

Case 2:
- When the image is running normally in RViz, disable the map.
- Reload the map in RViz.
- Sometimes image will freeze after the map reload.

**Current Solution**
- If the image freezes, disabling and re-enabling the image display in RViz usually restores it.
- However, if using the traffic light function, simply disable and re-enabling the image display does not work. Instead, switching the Reliability Policy (QoS) of the image is required to resume the image stream.

SuperChen0106, 2025-10-05:
Hi, I encountered a similar problem, but the above solution did not work.I am following this documentation (https://autoware-carla-launch.readthedocs.io/en/latest/scenarios/autoware.html) for the one vehicle scenario.The RViz2 Image panel shows "No Image". 
[image]
**Detailed Log Analysis & Root Cause:**
1. **No Camera Image Displayed**
The `autoware.log` shows that RViz2 and the zenoh-bridge-ros2dds are correctly set up to receive the image stream, but the data is likely never published from the source.
```
[rviz2-83] QImage::scaled: Image is a null image
[INFO] ... [zenoh_plugin_ros2dds::routes_mgr]: Route Subscriber (Zenoh:v1/sensing/camera/traffic_light/image_raw -> ROS:/sensing/camera/traffic_light/image_raw) created
```
2. **Critical QoS Mismatch Blocks Sensor Data**
My `autoware.log`shows fatal warnings indicating that the data stream for point clouds is being dropped due to incompatible QoS settings.
```
[carla_pointcloud_node-84] [WARN] ... requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] ... [sensing.lidar.concatenate_data]: New subscription discovered on topic '/sensing/lidar/concatenated/pointcloud', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```
3. **System Fails to Initialize Localization**
As a direct consequence of the missing sensor data, the entire system is stuck in an uninitialized state.
- **No Pose Available**: The planning modules are waiting indefinitely for the vehicle's position, as shown by this recurring log message:
`[scenario_selector-26] [INFO] ...: Waiting for current pose.`

**Attempted Solutions:**
Following the advice there, I tried to disable and re-enable the image display and change the Reliability Policy in RViz2 from Reliable to Best Effort for Image displays. However, this had no effect.Here are the complete autoware.log and zenoh_bridge_ros2dds.log files from my launch attempt, which contain the detailed error messages mentioned above.

[autoware.log](https://github.com/user-attachments/files/22708711/autoware.log)

[zenoh_bridge_ros2dds.log](https://github.com/user-attachments/files/22708714/zenoh_bridge_ros2dds.log)

SuperChen0106, 2025-10-05:
Thank you for your prompt reply!

[bridge.log](https://github.com/user-attachments/files/22710040/bridge.log)

[vehicle.log](https://github.com/user-attachments/files/22710041/vehicle.log)

habby1012, 2025-10-06:
Hi @SuperChen0106 !

Regarding the initial localization, it actually looks like your vehicle has already been initialized successfully in the video you shared.
However, users cannot manually specify the initial position, because the system automatically assigns a random starting pose when it first launches.

As for setting a goal pose, please make sure the arrow direction in RViz matches the driving direction of the lane.
In your video, the goal arrow was drawn in the opposite direction of the lane, which causes the planner to reject it.
Try setting the arrow in the same direction as traffic (keep right) — that should work! 😆

Also, if the image panel in RViz appears frozen right after startup, simply toggle the “Image” checkbox off and on again — that usually fixes it.

Please try these steps. If they work, everything should run smoothly afterward!