evshary/autoware_carla_launch #163: Auto button is not Enabled in RViZ when run carla with multiple autowares

lglokeshgandham, 2025-05-13:
Hi @evshary ,
Im able to run Carla with one Autoware successfully,
I want to run carla with multiple Autowares,

I'm running the setup in single machine ubuntu 22.04
and cloned humble branch

Steps to reproduce :
```
 git clone https://github.com/evshary/autoware_carla_launch.git -b humble
```
 Terminal1:
 ```
 	./CarlaUE4.sh -quality-level=Epic -world-port=2000 -RenderOffScreen -prefernvidia
```
 Terminal2:
 ```
 	./container/run-bridge-docker.sh
 	$cd autoware_carla_launch
	 $source env.sh
 	$./script/run-bridge-two-vehicles.sh
```
 Terminal3:
 ```
 	./container/run-autoware-docker.sh
 	$cd autoware_carla_launch
 	$source env.sh
 	$./script/run-autoware.sh v1 <IP>
 ```
 Terminal4:
 ```
 	./container/run-autoware-docker.sh
 	$cd autoware_carla_launch
 	$source env.sh
 	$./script/run-autoware.sh v2 <IP>
```
Observations:
I can see the Images displayed on both the Rviz
	
 For both The vehicles,Whenever I set GoalPose, I observe that the green path comes,but auto button is not enabled 
 
 after setting GoalPose, I see below continuous errors for v1 and v2 in terminal
 v1
 ```
 [component_container_mt-28] [1;33mCould not find valid vertex for ear clipping triangulation. Triangulation result might be invalid[0m
 ```
v2
```
 [ERROR] [1747100889.292781416] [localization.twist_estimator.gyro_odometer]: Vehicle twist msg is timeout. vehicle_twist_dt: 0.200000003[sec]
```
Attached the complete launch.log for v1, v2  bridge log files and screenshots

Can you please guide 

Thanks,
Lokesh

[v1_launch.log](https://github.com/user-attachments/files/20180259/v1_launch.log)
[v2_launch.log](https://github.com/user-attachments/files/20180258/v2_launch.log)
[bridge.log](https://github.com/user-attachments/files/20180257/bridge.log)
[vehicle1.log](https://github.com/user-attachments/files/20180260/vehicle1.log)
[vehicle2.log](https://github.com/user-attachments/files/20180261/vehicle2.log)
[image]
[image]

evshary, 2025-05-14:
Hi @lglokeshgandham this might be related to #150 
Could you please try the solution [here](https://github.com/evshary/autoware_carla_launch/issues/150#issuecomment-2678618138)?

lglokeshgandham, 2025-05-15:
Hello @evshary ,
Thankyou for the response,as mentioned,reducing the warn_rate resolved the issue

> Hi [@lglokeshgandham](https://github.com/lglokeshgandham) this might be related to [#150](https://github.com/evshary/autoware_carla_launch/issues/150) Could you please try the solution [here](https://github.com/evshary/autoware_carla_launch/issues/150#issuecomment-2678618138)?

evshary, 2025-08-05:
Close as completed

closed 2025-08-05 by evshary